Dasharo/open-source-firmware-validation #254: Debian not detected in scripts/secure-boot/generate-images/sb-img-wrapper.sh

pkubaj, 2024-03-29:
### Device

APU2

### RTE version

1.1.0

### Brief summary

Debian not detected in scripts/secure-boot/generate-images/sb-img-wrapper.sh

### How to reproduce

```
debian@debian:~/open-source-firmware-validation/scripts/secure-boot/generate-images$ ./sb-img-wrapper.sh 
```

### Expected behavior

Debian should be detected.

### Actual behavior

```
Creating image for INTERMEDIATE 

Installing dependencies
Unknown Package Manager
debian@debian:~/open-source-firmware-validation/scripts/secure-boot/generate-images$ lsb_release -is
No LSB modules are available.
Debian
```

### Link to screenshots or logs

There are no logs.

macpijan, 2024-05-10:
Related to: https://github.com/Dasharo/open-source-firmware-validation/issues/256

artur-rs, 2024-10-24:
PR with fixes has been merged, closing

closed 2024-10-24 by artur-rs