Dasharo/open-source-firmware-validation #255: Unable to locate package sbsigntools

pkubaj, 2024-03-29:
### Device

APU2

### RTE version

1.1.0

### Brief summary

Unable to locate package sbsigntools when running sb-img-wrapper.sh

### How to reproduce

`./sb-img-wrapper.sh`

### Expected behavior

Tools should be installed and script should proceed.

### Actual behavior

```
Creating image for INTERMEDIATE 

Installing dependencies
Detected Package Manager: apt
Detected Distro: Debian
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package sbsigntools
Cannot install sbsigntools faketime : (100)
```

### Link to screenshots or logs

There ane no logs.

pkubaj, 2024-03-29:
This issue happens when I force using apt in `generate-image.sh` as otherwise Debian is not detected.

pkubaj, 2024-03-29:
Package name is changed to `sbsigntool`.

BeataZdunczyk, 2024-03-29:
> This issue happens when I force using apt in generate-image.sh as otherwise Debian is not detected.

_Debian_? Do you mean Ubuntu @pkubaj?

pkubaj, 2024-04-02:
No, Debian, since that's what's installed on the VM.

macpijan, 2024-05-10:
Related to: https://github.com/Dasharo/open-source-firmware-validation/issues/256

closed 2024-10-24 by artur-rs