Dasharo/open-source-firmware-validation #257: Conne

macpijan, 2024-04-02:
**Test environment version**
* Branch: 
* Commit: 

**Dasharo version (if applicable)**

**Dasharo variant (if applicable)**

**Affected component(s) or functionality (if applicable)**

**Brief summary**

Implement DRY strategy.
Avoid reimplementing low-level flashing and platform control login in multiple places:
- test repo
- CLI repo
- flashing scripts on the platform

Keep the low-level details only in one place, and provide an interface to be used in tests.

**Additional context**

macpijan, 2024-04-16:
@PLangowski 

I have tried various way of sharing this, and probably the best way is as in
the linked branches. I would start from there adn check if we can call
basic functions (like flash probe/read) from osfv-script repo in all these cases:
- from osfv_cli in poetry shell: https://github.com/Dasharo/osfv-scripts/tree/main/osfv_cli#development
- from osfv_cli when installed through Make install
- from robot tests, through the RobotRTE library (can be check on any of the available platforms)

If the answer is yes, and we cannot see probles with this approach, we can continue with
cleanup, further improvements, and replacing the previous Rte library with the calls
from osfv-scripts libs.

High-level TODOs:
- replace calls to https://github.com/3mdeb/rtectrl-rest-api with these to
  osfv-scripts libs 
- replace calls to https://github.com/3mdeb/sonoff-rest-ap with these to
  osfv-scripts libs
- drop the low-level hardware access procedures from open-source-firmware-validation
  and replace them with higher-level calls from osfv-scripts; especially the flashing,
  perhaps also the Power On, and similar
- use snipeit from osfv-scripts package, not imported from path like here:
  https://github.com/Dasharo/open-source-firmware-validation/commit/c08c808be6878e462587bb940a42bdb9c0a4f6e0#diff-2bd69584d51e03632b6b478cae660b9d98acb614bedfcf8ad005f876c9d6744eR528
- in the open-source-firmware-validation repo we've had the `-v snipeit:no` switch
  which allowed us to run the scripts without snipeit; this should still be somehow
  supported, but the snipeit calls are somehow convoluted in the osfv-scripts
  already; we need a proposal on how to handle that
- add osfv-scripts to requirements.txt to make sure it gets instaleld into venv

PLangowski, 2024-04-18:
@macpijan 
I've heard the Sonoff API has changed. It seems to be confirmed when  i try to use `osfv_cli` to control sonoff
```
osfv_cli sonoff --rte_ip 192.168.10.107 get
Getting Sonoff relay state...
Failed to get Sonoff relay state. Error: 404 Client Error: Not Found for url: http://192.168.10.170/switch/sonoff_s20_relay
```

Below is the content of the `sonoff.sh` script on the rte in question:

```
#!/bin/bash

SONOFF_IP=192.168.10.170

if [[ "$1" == "off" ]]; then
    wget -q -O - http://$SONOFF_IP/cm?cmnd=Power%20off --method=POST
    echo -e '\n'
elif [[ "$1" == "on" ]]; then
    wget -q -O - http://$SONOFF_IP/cm?cmnd=Power%20On --method=POST
    echo -e '\n'
elif [[ "$1" == "toggle" ]]; then
    wget -q -O - http://$SONOFF_IP/cm?cmnd=Power%20TOGGLE --method=POST
    echo -e '\n'
elif [[ "$1" == "show" ]]; then
    wget -q -O - http://$SONOFF_IP/cm?cmnd=Power
    echo -e '\n'
else
    echo -e "\$1 == on|off|show|toggle\nEdit this script to set the sonoff ip."
    echo -e 'Current state:'
    wget -q -O - http://$SONOFF_IP/cm?cmnd=Power
    echo -e '\n'
fi
```
These are clearly different URLs.
Is anyone working on adapting `osfv_cli` to the new API?

macpijan, 2024-04-18:
It is adapated already. Which versino are you using?

PLangowski, 2024-04-18:
main, tag v0.4.2

macpijan, 2024-04-18:
Ok it is here: https://github.com/Dasharo/osfv-scripts/pull/11/files#diff-152929c4b9e899cc69afdcfad33041262ac6bdc8634e5175bfb43cb2962d1bdf

PLangowski, 2024-04-18:
@macpijan both PRs linked to this issue are ready for review. I have tested `Flash Read` on msi-pro-z690-a-ddr4 and minnowboard turbot. Everything seems to work, including running tests without snipeit (that requires sonoff and pikvm ips to be specified in command line). Here's how to run a test that way:

```
robot -L TRACE -v snipeit:'no' -v sonoff_ip:192.168.10.170 -v pikvm_ip:192.168.10.99 -v rte_ip:192.168.10.107 -v config:"msi-pro-z690-a-wifi-ddr4" -t "TEST  /path/to/suite
```

closed 2024-06-11 by macpijan