Dasharo/open-source-firmware-validation #263: [ ERROR ] Error in file '/home/debian/open-source-firmware-validation/dasharo-compatibility/dcu.robot' on line 9: Importing library 'FakerLibrary' failed: ModuleNotFoundError: No module named 'FakerLibrary'

pkubaj, 2024-04-10:
### Device

VP2410

### RTE version

1.1.0

### Brief summary

DCU tests fail to start

### How to reproduce

`robot -b command_log.txt -v snipeit:no -L TRACE -v rte_ip:192.168.10.233 -v device_ip:192.168.10.24 -v config:protectli-vp2410 dasharo-compatibility/dcu.robot `

### Expected behavior

Tests should start.

### Actual behavior

```
[ ERROR ] Error in file '/home/debian/open-source-firmware-validation/dasharo-compatibility/dcu.robot' on line 9: Importing library 'FakerLibrary' failed: ModuleNotFoundError: No module named 'FakerLibrary'
Traceback (most recent call last):
  None
PYTHONPATH:
  /home/debian/open-source-firmware-validation/venv/bin
  /usr/lib/python311.zip
  /usr/lib/python3.11
  /usr/lib/python3.11/lib-dynload
  /home/debian/open-source-firmware-validation/venv/lib/python3.11/site-packages
==============================================================================
```

### Link to screenshots or logs

[command_log.txt](https://github.com/Dasharo/open-source-firmware-validation/files/14932452/command_log.txt)

macpijan, 2024-04-10:
you need to rebuild venv if dependenics gets updated: https://github.com/Dasharo/open-source-firmware-validation/blob/develop/requirements.txt#L19

pkubaj, 2024-04-10:
Ah, I didn't notice that, thanks.

closed 2024-04-10 by pkubaj