Dasharo/open-source-firmware-validation #264: STB001.002 fails with The device has been reset during the test! when it's not

pkubaj, 2024-04-11:
### Device

VP2410

### RTE version

1.1.0

### Brief summary

STB001.002 fails with The device has been reset during the test! when it's not

### How to reproduce

`robot -b command_log.txt -v snipeit:no -L TRACE -v rte_ip:192.168.10.233 -v device_ip:192.168.10.24 -v config:protectli-vp2410 -v fw_file:protectli_vp2410_v1.1.0.rom -t STB001.002* dasharo-performance/platform-stability.robot`

### Expected behavior

Test should run.

### Actual behavior

```
STB001.002 Verify if no reboot occurs in the OS (Ubuntu 22.04) :: ... | FAIL |
The device has been reset during the test!
```

### Link to screenshots or logs

[logs.tar.gz](https://github.com/Dasharo/open-source-firmware-validation/files/14948431/logs.tar.gz)

macpijan, 2024-05-10:
Sounds like needs reproduction and analysis/fix.

macpijan, 2024-05-10:
@pkubaj please check for duplicates before opening new issue: https://github.com/Dasharo/open-source-firmware-validation/issues/233
It is enough to report with new conditions (like a device) in the old ticket.