Dasharo/open-source-firmware-validation #265: Execute Command In Terminal not working in UEFI Shell

PLangowski, 2024-04-12:
### Device

Minnowboard Turbot

### RTE version

0.7.5

### Brief summary

`Execute Command in Terminal` fails in UEFI Shell.

### How to reproduce

`robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.112 -v config:minnowboard-turbot  assessment/txe-secure-boot.robot`

### Expected behavior

The keyword should execute the command and return the output.

### Actual behavior

The keyword fails -  for some reason it waits for an empty string instead instead of prompt.

### Link to screenshots or logs

[image]

### Additional context

As of now, executing commands in UEFI Shell requires running the following 3 keywords:
```
Write Bare Into Terminal    <command>
Press Enter
${out}=    Read From Terminal Until Prompt
```
A new keyword should be defined, which achieves the above.

macpijan, 2024-05-28:
We enter into UEFI shell in these cases:

```
./dasharo-security/secure-boot.robot:150:    Enter UEFI Shell
./dasharo-security/secure-boot.robot:167:    Enter UEFI Shell
./dasharo-security/secure-boot.robot:184:    Enter UEFI Shell
./dasharo-security/secure-boot.robot:226:    Enter UEFI Shell
./dasharo-security/secure-boot.robot:238:    Enter UEFI Shell
./lib/secure-boot-lib.robot:219:Enter UEFI Shell
./assessment/txe-secure-boot.robot:35:    Enter UEFI Shell
```

But it seems we only execute some commands in shell in the last two, so only these should be fixed.

Steps to solve this:
- [ ] Create new file under self-tests/terminal.robot
  - [ ] Test should enter UEFI shell and execute some command (even help or anything), reading it's output
  - Try to do this via `Execute Commanad in Terminal` and via workaround proposed in the issue, to understand the problem
- [ ] Improve the `Execute Command In Terminal` to properly handle it, or introduce a new keyword for executing command in UEFI shell, and place it under lib/terminal.robot
- [ ] Replace all occurencies with a new kwd

JanPrusinowski, 2024-05-29:
Fix availible at: https://github.com/Dasharo/open-source-firmware-validation/pull/302

krystian-hebel, 2024-10-07:
This hasn't been properly fixed. The new keyword reads exactly 1s worth of output, which in some cases isn't enough:
[log.zip](https://github.com/user-attachments/files/17281357/log.zip). Even when the keywords depending on the output of this command pass, it is barely enough to get the line we're interested in, see output of `Execute UEFI Shell Command smbiosview -t 0` started at `16:44:31.248`.

The keyword should either return all of the command's output (up until timeout), or not consume any of it.

philipandag, 2024-12-16:
The fix is continued at https://github.com/Dasharo/open-source-firmware-validation/pull/528

closed 2025-04-15 by BeataZdunczyk