Dasharo/open-source-firmware-validation #269: Introduce headset flag

macpijan, 2024-04-18:
The "AUD004.001 External headset recognition (Ubuntu 20.04)" is always executed if audio is tested. But some platforms (like the VP2410) do not have audio jacks for headset to be tested. So running them makes no sense.

PLangowski, 2024-04-25:
@macpijan Could you list the platforms that do/do not support external headsets so that I can modify their configs?

PLangowski, 2024-04-26:
As of now, I added the flag, which is false by default. I set it to true for some of the platforms. The test passes for MSI PRO Z690-A DDR4, for which the flag is set to true. It also skips the test for Protectli VP4630, where the flag is false.

macpijan, 2024-04-26:
I am sure you can figure it out by yourself. If in trouble, pleas ask specific questions.

https://eu.protectli.com/product-comparison/

PLangowski, 2024-04-26:
@macpijan
I checked the platforms and modified the configs accordingly.

macpijan, 2024-05-28:
Fixed in: https://github.com/Dasharo/open-source-firmware-validation/pull/277

closed 2024-05-28 by macpijan