Dasharo/open-source-firmware-validation #282: "Boot Dasharo Tools Suite" fails on Novacusom laptop

m-iwanicki, 2024-04-30:
### Device

novacustom-ns50mu

### RTE version

0.7.4-rc4

### Brief summary

Keyword waits for "Enter an option:" until timeout

### How reproducible

Always

### How to reproduce

Run `Boot Dasharo Tools Suite` keyword on laptop with `DUT_CONNECTION_METHOD==pikvm`.

### Expected behavior

Keyword should succeed

### Actual behavior

Keyword is stuck on [Read From Terminal Until](https://github.com/Dasharo/open-source-firmware-validation/blob/62af487ea921d2a8f1ca1a29cfd56b1a588dde51/lib/dts-lib.robot#L47) keyword because there is no serial output from DTS.

### Link to screenshots or logs

Output on serial ends at ```Booting `boot'```

```
12:24:17.818 	TRACE 	Arguments: [ 'Enter an option:' ] 	
12:24:19.532 	TRACE 	recv b'PiKVM Composite KVM Device \x1b[1m\x1b[37m\x1b[44m\x1b[12;24H ' 	
12:24:19.532 	TRACE 	recv b' Samsung SSD 980 PRO 250GB \x1b[0m\x1b[37m\x1b[40m\x1b[1m\x1b[33m' 	
12:24:19.532 	TRACE 	recv b'\x1b[44m\x1b[28D>\x1b[01CSamsung SSD ' 	
12:24:19.534 	TRACE 	recv b'9' 	
12:24:19.802 	TRACE 	recv b'80 PRO 250GB \x1b[1m\x1b[37m\x1b[44m\x1b[11;24H  PiKVM Composi' 	
12:24:19.802 	TRACE 	recv b'te KVM Device \x1b[0m\x1b[37m\x1b[40m\x1b[1m\x1b[33m\x1b[44m\x1b[29D>\x1b[' 	
12:24:19.802 	TRACE 	recv b'01CPiKVM Composite KVM Device' 	
12:24:20.201 	TRACE 	recv b' \x1b[1m\x1b[37m\x1b[44m\x1b[12;24H  Samsung SSD 980 PRO 250GB' 	
12:24:20.201 	TRACE 	recv b' \x1b[0m\x1b[37m\x1b[40' 	
12:24:20.214 	TRACE 	recv b'm\x1b[2J\x1b[01;01H\x1b[0m\x1b[30m\x1b[47mWelcome to GRUB!\n\r\r\n\r' 	
12:24:20.267 	TRACE 	recv b'\r\x1b[0m\x1b[37m\x1b[40merror: no such device: ((hd1,msdos1' 	
12:24:20.267 	TRACE 	recv b')/EFI/BOOT)/EFI/BOOT/grub.cfg.\n\r' 	
12:24:20.289 	TRACE 	recv b"\rerror: serial port `com0' isn't found.\n\r" 	
12:24:20.332 	TRACE 	recv b'\r\x1b[0m\x1b[30m\x1b[40m\x1b[2J\x1b[01;01H\x1b[0m\x1b[37m\x1b[40m  Booting' 	
12:24:20.332 	TRACE 	recv b" `boot'\n\r\r\n\r"
```

macpijan, 2024-05-10:
It should work as described here: https://github.com/Dasharo/open-source-firmware-validation/blob/main/lib/dts-lib.robot#L25

If it does not work, there indeed might be a problem and we need to take a closer look into it.
ATM we should check from `develop` branch - it is not clear which repo revision was used - please clarify this in future reports.

macpijan, 2024-05-10:
This is a duplicate of: https://github.com/Dasharo/open-source-firmware-validation/issues/196

@m-iwanicki please search existing issues before creating a new one.

closed 2024-05-10 by macpijan