Dasharo/open-source-firmware-validation #303: Move default values for login and password from PiKVM keywords to seprate keys

JanPrusinowski, 2024-06-03:
**Test environment version**
* Branch:  develop
* Commit: 

**Dasharo version (if applicable)**

**Dasharo variant (if applicable)**

**Affected component(s) or functionality (if applicable)**

pikvm-rest-api

**Brief summary**
PiKVM uses values stored as defaults in keywords to login to the device.

**Additional context**
/pikvm-rest-api/pikvm-comm.robot

```
Write Bare PiKVM
    [Documentation]    Keyword for PiKVM writing bare into Terminal.
    ...    WARNING: Supports only small characters.
    ...    text - text to be written,
    ...    pikvm_ip - IP of the piKVM to send key input,
    ...    login - piKVM login,
    ...    password - piKVM password
    [Arguments]    ${text}    ${login}=admin    ${password}=admin
    Write Text PiKVM    ${text}    ${PIKVM_IP}    ${login}    ${password}
```

Keywords use default `'=admin'` values for login and password - this makes finding of this key very hard.

Refactor the code by creating a new key stored with other keys and link it to this keywords.

This issue may apply to other login methods too.

macpijan, 2024-07-16:
@JanPrusinowski I think there was a PR from you on this, but it's not been linked here. Please do this, and close if merged.

macpijan, 2024-08-14:
@JanPrusinowski ping

JanPrusinowski, 2024-08-14:
Was merged at: https://github.com/3mdeb/pikvm-rest-api/pull/13

closed 2024-08-14 by JanPrusinowski