Dasharo/open-source-firmware-validation #313: Support Windows cases with DCU OPTIONS_LIB

macpijan, 2024-06-10:
**Test environment version**
This MR: https://github.com/Dasharo/open-source-firmware-validation/pull/305

**Dasharo version (if applicable)**
N/A

**Dasharo variant (if applicable)**
NCM Laptops

**Affected component(s) or functionality (if applicable)**
Executing tests with DCU instead of serial for changing firmware options

**Brief summary**
In the linked MR, we have some proposal with following limitations:
1. We assume that ubuntu is the default OS to boot
2. We do not support Windows at all, while have some relevant number of Windows test cases

**Additional context**

macpijan, 2024-07-16:
@BeataZdunczyk That may require some more though into how we could support windows cases with this approach.

JanPrusinowski, 2024-09-11:
Because of the way that DCU works each machine is set to boot automatically to Ubuntu. Usually to change onto which OS we boot it is required to enter BIOS and select the option - which is impossible for machines using DCU as they do not allow for serial console output in any way. Flashing FW each time when we want to boot to different OS is sub optimal solution as it would take a long time and cause wear in BIOS chip.

But there is a way to set the next reboot target from Ubuntu using `efibootmgr`. This way instead of picking the OS from the BIOS boot menu it can be done from Ubuntu. DUT would have to boot to Ubuntu - set the next boot target and reboot.
Default boot target would stay as Ubuntu - so that changing the boot device to invalid one would not brick the device.

This way we can introduce not only support for Windows, but all other OS. Only requirement is that the OS uses SSH for connection.

mkopec, 2024-09-11:
Seems like a good idea to me. Setting `BootNext` in efibootmgr should work well enough here.