Dasharo/open-source-firmware-validation #321: `PXE004.001 DTS option is available and works correctly` checks more than it should

krystian-hebel, 2024-06-20:
Even though the name suggests that booting to DTS should be enough, the test actually checks if DTS knows about Dasharo firmware for given platform. This isn't true when doing tests before initial releases. There are dedicated tests for DTS that check for availability of updates, and lack of them doesn't mean that there is a problem with network boot.

tym2k1, 2024-06-26:
This is how DTS looks on unsupported device:

[image]

This is DTS on QEMU from `develop` branch of [OSFV](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/README.md?plain=1#L203-L213)

I see two solutions:

- We can disable the check for Install/Update entirely:

    ```diff
     PXE004.001 DTS option is available and works correctly
         [Documentation]    This test aims to verify that the Dasharo Tools Suite
         ...    option in Dasharo Network Boot Menu allows booting into DTS.
         Skip If    not ${TESTS_IN_FIRMWARE_SUPPORT}    PXE004.001 not supported
         Power On
         ${boot_menu}=    Enter Boot Menu Tianocore And Return Construction
         Enter Submenu From Snapshot    ${boot_menu}    ${IPXE_BOOT_ENTRY}
         ${ipxe_menu}=    Get IPXE Boot Menu Construction
         Enter Submenu From Snapshot    ${ipxe_menu}    Dasharo Tools Suite

         Set DUT Response Timeout    5m
         ${out}=    Read From Terminal Until    Enter an option
         Should Contain    ${out}    Dasharo HCL report
    -    # TODO:
    -    # On some targets (such as QEMU), we get option to install, not to update
    -    Should Contain Any    ${out}    Install Dasharo firmware    Update Dasharo firmware
         Should Contain    ${out}    Load your DES keys
         Should Contain    ${out}    Start SSH server
         Should Contain    ${out}    Shell
    ```

- We can add an option that it could containt the `Restore firmware from Dasharo HCL report` line (or other line specific to the scenario where Dasharo is not supported on the platform)

    ```diff
     PXE004.001 DTS option is available and works correctly
         [Documentation]    This test aims to verify that the Dasharo Tools Suite
         ...    option in Dasharo Network Boot Menu allows booting into DTS.
         Skip If    not ${TESTS_IN_FIRMWARE_SUPPORT}    PXE004.001 not supported
         Power On
         ${boot_menu}=    Enter Boot Menu Tianocore And Return Construction
         Enter Submenu From Snapshot    ${boot_menu}    ${IPXE_BOOT_ENTRY}
         ${ipxe_menu}=    Get IPXE Boot Menu Construction
         Enter Submenu From Snapshot    ${ipxe_menu}    Dasharo Tools Suite

         Set DUT Response Timeout    5m
         ${out}=    Read From Terminal Until    Enter an option
         Should Contain    ${out}    Dasharo HCL report
         # TODO:
         # On some targets (such as QEMU), we get option to install, not to update
    -    Should Contain Any    ${out}    Install Dasharo firmware    Update Dasharo firmware
    +    Should Contain Any    ${out}    Install Dasharo firmware    Update Dasharo firmware    Restore firmware from Dasharo HCL report
         Should Contain    ${out}    Load your DES keys
         Should Contain    ${out}    Start SSH server
         Should Contain    ${out}    Shell
    ```

I'm more in favour of the first option as it's more elegant

krystian-hebel, 2024-07-17:
Fix was merged, closing the issue.