Dasharo/open-source-firmware-validation #336: Custom fan curve test fails

filipleple, 2024-06-27:
### Device

Novacustom v560TU

### RTE version

--

### OSFV version

--

### Affected component(s) or functionality

Custom fan curve test

### Brief summary

Custom fan curve test fails due to a timeout running `sensors`

### How reproducible

100%

### How to reproduce

Try to run the Custom fan curve test

### Expected behavior

It should pass

### Actual behavior

```
RuntimeError: No match found for 'root@ubuntu-V54x-6x-TU:/home/ubuntu#' in 30 seconds
Output:

# sensors-detect version 3.6.0
# System: Notebook V54x_6x_TU [V54x_6x_TU] (laptop)
# Kernel: 6.9.0-060900rc7-generic x86_64
# Processor: Intel(R) Core(TM) Ultra 7 155H (6/170/4)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No

(...)

````

### Link to screenshots or logs

Log pasted above

mkopec, 2024-07-02:
> RuntimeError: No match found for 'root@ubuntu-V54x-6x-TU:/home/ubuntu#' in 30 seconds

Hmm, I guess lm-sensors may be waiting for input when doing initial scanning. You could prepend `yes '\n' | ` before the `sensors` command to work around it I suppose...

...but the tests won't work on novacustom anyway, we don't have an OS driver to expose the fan speed

macpijan, 2024-07-16:
> ...but the tests won't work on novacustom anyway, we don't have an OS driver to expose the fan speed
@filipleple @mkopec 
Did we remove the support flag from config then?

philipandag, 2024-07-19:
> ...but the tests won't work on novacustom anyway, we don't have an OS driver to expose the fan speed
Is the test faulty then or is it just a matter of disabling it for the novacustom?

philipandag, 2024-07-19:
Working on the issue here:
https://github.com/Dasharo/open-source-firmware-validation/pull/394

philipandag, 2024-07-22:
It seems that the fan tests should just not be supported for novacustom. No pwm devices are visible. At least on the v560tu and v560tdm on which I have tested. @mkopec  do you know, is it all novacustom laptops or only the v5x family that don't have a driver at this moment?

philipandag, 2024-07-22:
I have tested that an NS50MU laptop behaves in the same way, so my fix for the moment will be updating nova-custom-common config to not support fan tests.

mkopec, 2024-07-25:
This issue is blocked by https://github.com/Dasharo/dasharo-issues/issues/820