Dasharo/open-source-firmware-validation #339: Properly implement and use "Stop Logging To Terminal" kwd

macpijan, 2024-06-28:
When running tests over serial, we sometimes can get unwanted random errors from dmesg in between the output we need.
This should be disabled early in the boot process, or even on the OS preparation stage, so we do not get unwanted outputs, breaking tests randomly.

Some approach has been done here: https://github.com/Dasharo/open-source-firmware-validation/pull/286/files#diff-9c5722de9fc8a63153bcc0396f26d4409dce4c9a70c217e14123b20693e481fcL44

macpijan, 2024-07-16:
I think on Ubuntu it needed /etc/sysctl change as printk from kernel command line is overriden by this

macpijan, 2024-07-16:
@BeataZdunczyk Ideally this is integrated into installers, in tests maybe it can be verified if the OS was installed correctly.

philipandag, 2024-08-09:
@macpijan So we would like to disable these logs when installing the OS, not only for the sake of a test case -- using a KW? 

- Where can we make a change? In [Dasharo preseeds](https://github.com/Dasharo/preseeds/tree/main/ubuntu)? Or maybe add a procedure of permamently disabling the unwanted logs to the post installation guides in docs?

- What was wrong in the approach by @mkopec? 
  > Stop Logging To Terminal
    [Documentation]    This keyword stops all unwanted logging to the terminal
    ...    we're using for the shell. This prevents garbage from being logged.
    Execute Linux Command
    ...    sed -i " s/'.*ForwardToConsole[::space::]*=.*'/ForwardToConsole=no/g" /etc/systemd/journald.conf
    Execute Linux Command    echo 0 > /proc/sys/kernel/printk
    Execute Linux Command    systemctl restart systemd-journald

  Is it that we want the change to be permament and applied during installation, not on per-test basis?
   
- >  in tests maybe it can be verified if the OS was installed (configured) correctly.

  By that, do you mean making a KW for asserting that the OS was installed properly (the logs are disabled) and making the test fail if that is not the case?

macpijan, 2024-08-09:
> What was wrong in the approach by @mkopec?

I think the main problem is it does not work, as whatever we set to `/proc/sys/kernel/printk` on Ubuntu is overriden with:

```
cat /etc/sysctl.d/10-console-messages.conf

# the following stops low-level messages on console
kernel.printk = 1 4 1 7
```

So we need to edit this file in installer, and in the https://github.com/Dasharo/open-source-firmware-validation/blob/develop/util/basic-platform-setup.robot#L129 file we can add a test for checking ubuntu configuration. One of the checks would be to check the proper `/proc/sys/kernel/printk` setup.

philipandag, 2024-08-12:
Thanks for the tips. I think the prints may also be disabled by uncommenting this line in /etc/sysctl.conf:

```
# Uncomment the following to stop low-level messages on console
#kernel.printk = 3 4 1 3
```

I will first try to change this setting simply by appending new `kernel.printk` value to the end of `/etc/sysctl.conf` and then by overwriting the `/etc/sysctl.d/10-console-messages.conf`. 
Testing if this works properly should be as simple as checking the first digit in the output of `cat /proc/sys/kernel/printk`

philipandag, 2024-08-12:
We are already setting the kernel.printk in the preseeds:
```
in-target sed -i 's/kernel.printk.*/kernel.printk = 1 4 1 7/' /etc/sysctl.d/10-console-messages.conf;\
```

I suppose log level 1 is to high and we would like to lower it to 0? I think we only need to change it here.

philipandag, 2024-08-12:
https://github.com/Dasharo/preseeds/pull/19
https://github.com/Dasharo/open-source-firmware-validation/pull/449

philipandag, 2024-08-12:
@macpijan do you know a good way to reproduce the issue?

philipandag, 2024-09-12:
The PRs are merged https://github.com/Dasharo/open-source-firmware-validation/issues/339#issuecomment-2283506905, closing.

closed 2024-09-12 by philipandag