Dasharo/open-source-firmware-validation #342: CI randomly fails

tym2k1, 2024-07-01:
### Device

Doesn't apply

### RTE version

Doesn't apply

### OSFV version

branch: pxe004_001_only_check_if_boots; commit: 5fb5716eaf55

### Brief summary

CI randomly fails for PR's

### How to reproduce

Run CI/CD for new PR

### Expected behavior

CI workflows should provide the same output for the same commit in the [Start Keywords self-tests with QEMU](https://github.com/Dasharo/open-source-firmware-validation/actions/runs/9681493138/workflow#L43-L45)

### Actual behavior

Workflow [fails](https://github.com/Dasharo/open-source-firmware-validation/actions/runs/9681439904/job/26712030745?pr=332) randomly and [works](https://github.com/Dasharo/open-source-firmware-validation/pull/332/checks) after rerunning

### Link to screenshots or logs

[Link](https://github.com/Dasharo/open-source-firmware-validation/actions/runs/9681439904/job/26712030745?pr=332)

macpijan, 2024-07-16:
@BeataZdunczyk Needs checking which test/keywords randomly fails. Might be related to other labeled reliability issues.

macpijan, 2024-08-05:
We will be monitoring this issue by adding logs as artifacts: https://github.com/Dasharo/open-source-firmware-validation/pull/407#issuecomment-2268754938

So far, the failures are inconsistent, we need more data and logs have been missing so far.