Dasharo/open-source-firmware-validation #353: QEMU and talosII have the same cpuid

m-iwanicki, 2024-07-05:
### Device

QEMU

### RTE version

N/A

### OSFV version

develop

### Brief summary

Get Current CONFIG fails for QEMU due to same cpuid as Talos II

### How reproducible

Always

### How to reproduce

Run this test:

```shell
robot -v snipeit:no -L TRACE -v rte_ip:127.0.0.1 -v config:qemu test.robot
```

`test.robot`:

```robot
*** Settings ***
Library             Telnet
Library             SSHLibrary
Resource            variables.robot
Resource            keywords.robot
Suite Setup         Run Keyword
...                     Prepare Test Suite
Suite Teardown      Run Keyword
...                     Log Out And Close Connection

*** Test Cases ***
TEST
    ${conf}=    Get Current CONFIG    ${CONFIG_LIST}
    Log To Console    ${conf}
```

### Expected behavior

```==============================================================================
TEST                                                                  .[{'ip': '127.0.0.1', 'cpuid': '02c0014296737c0!', 'platform': 'qemu', 'platform_vendor': 'qemu'}]
TEST                                                                  | PASS |
------------------------------------------------------------------------------
Test                                                                  | PASS |
1 test, 1 passed, 0 failed
==============================================================================
```

### Actual behavior

```
==============================================================================
TEST                                                                  .[{'ip': '192.168.4.133', 'cpuid': '02c0014296737c0d', 'platform': 'talosII', 'platform_vendor': 'Raptor Engineering, LLC', 'sonoff_ip': '192.168.4.106'}]
TEST                                                                  | PASS |
------------------------------------------------------------------------------
Test                                                                  | PASS |
1 test, 1 passed, 0 failed
==============================================================================
```

### Link to screenshots or logs

[logs.tar.gz](https://github.com/user-attachments/files/16106894/logs.tar.gz)

### Solutions you've tried

Change QEMU cpuid to some unique value in `variables.robot`

macpijan, 2024-07-16:
@BeataZdunczyk We do not really need cpuid at all. As of https://github.com/Dasharo/open-source-firmware-validation/commit/caf414360c062ec19adc26fbf750bc5efae10c6e it is not necessary in this kwd as well. So we can use this issue to remove cpuid from variables.robot entirely, and see if it still works.

wiktormowinski, 2024-07-18:
something like that?: https://github.com/Dasharo/open-source-firmware-validation/pull/387
simply removing Talos' cpuid from variables.robot seems to be the solution

BeataZdunczyk, 2024-07-19:
> [wiktormowinski](https://github.com/wiktormowinski) closed this as [completed](https://github.com/Dasharo/open-source-firmware-validation/issues?q=is%3Aissue+is%3Aclosed+archived%3Afalse+reason%3Acompleted) [2 minutes ago](https://github.com/Dasharo/open-source-firmware-validation/issues/353#event-13572748243)

misclick? PR is not merged @wiktormowinski.

wiktormowinski, 2024-07-19:
oh ok, should i do it? @BeataZdunczyk

m-iwanicki, 2024-07-19:
@wiktormowinski I would say to wait until https://github.com/Dasharo/open-source-firmware-validation/pull/361 is merged as that PR fixes error from this PR.

wiktormowinski, 2024-07-19:
sure, got it

m-iwanicki, 2024-07-22:
Just reread my comment and to specify, wait with closing this issue. PR probably can be merged but should probably be accepted by @macpijan as it's his branch/PR you're merging into

closed 2024-07-22 by macpijan