Dasharo/open-source-firmware-validation #356: Tianocore Reset S

macpijan, 2024-07-05:
### Device

V1211

### RTE version

v1.0.0

### OSFV version

c16334e10441cbf9d73560eb5b0443de7d8aa46b (HEAD -> regression_protectli_v121x, origin/regression_protectli_v121x)

### Affected component(s) or functionality

menus . Tianocore Reset System

### Brief summary

System is not reset

### How to reproduce

Run SB suite.

```
DEVICE_IP=192.168.10.222 FW_FILE=protectli_v1211_v0.9.2.rom RTE_IP=192.168.10.227 CONFIG=protectli-v1211 SNIPEIT_NO=true ./scripts/run.sh dasharo-security/secure-boot.robot 
```

### Expected behavior

System is reset

### Actual behavior

System is not reset

 Either SBO002.001 or SB002.002 or both, sometimes fail.

### Link to screenshots or logs

Noticed a few times especially in the SB suite:

[image]

macpijan, 2024-07-16:
@BeataZdunczyk Need for a reliability test for this keyword to ensure it works as expected.

JanPrusinowski, 2024-07-24:
Fixed in: https://github.com/Dasharo/open-source-firmware-validation/pull/406
SSH connection method wasnt always changed back to default DUT connection method.
[log.zip](https://github.com/user-attachments/files/16361348/log.zip)

macpijan, 2024-07-26:
I am not so convinced that this is the root cause of the problem here...
Please note that we have entered some menu, in BIOS, changes some settings, tried to save & exit, and only then it failed (it did not reset).

If it was the problem, we would not eneter in any of the menus in the first place.

JanPrusinowski, 2024-08-09:
I have managed to replicate the issue on the current develop after 5th run.
[secure-boot.robot_log.zip](https://github.com/user-attachments/files/16560315/secure-boot.robot_log.zip)

JanPrusinowski, 2024-09-10:
I have found an issue which might have an impact on stability of the tests: https://github.com/Dasharo/open-source-firmware-validation/issues/496

JanPrusinowski, 2024-10-08:
I have reworked the Tianocore Reset System kwd. And run this kwd 50 times. 
[log.zip](https://github.com/user-attachments/files/17289470/log.zip)