Dasharo/open-source-firmware-validation #357: Wrong PASS criteria in Verified Boot

filipleple, 2024-07-05:
### Device

any

### RTE version

any

### OSFV version

`a6145514ee111e264aef873551b2bbc473b9626a`

### Affected component(s) or functionality

[dasharo-security/verified-boot.robot](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/dasharo-security/verified-boot.robot)

### Brief summary

Verified Boot suite fails due to wrong assumptions

### How reproducible

100%

### How to reproduce

Run dasharo-security/verified-boot.robot

### Expected behavior

It should PASS when a recovery popup appears after resigning FW with own keys and flashing to `RW_SECTION_*` - IIRC that's how verified boot should work, since the keys will be different than the initial build ones

### Actual behavior

It FAILS - expects booting from Slot A

### Link to screenshots or logs

[VBO012.001](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/dasharo-security/verified-boot.robot#L144)

macpijan, 2024-07-16:
@BeataZdunczyk adding vboot label

BeataZdunczyk, 2024-07-19:
Isn't this fixed, @JanPrusinowski? Please link the PR that fixes this issue and share the platform on which it was successfully tested.

JanPrusinowski, 2024-07-19:
I was unable to reproduce this bug. @filipleple could you provide more info about the platform, and which particular test is the issue?

filipleple, 2024-07-19:
@JanPrusinowski it was the NovaCustom V560TU, tests [VBO001.001 ... VBO003.001](https://docs.dasharo.com/unified-test-documentation/dasharo-security/201-verified-boot). It's important to flash both RW_SECTION_A and B, and not full BIOS region

EDIT: These three are manual only tests, they're not automated. Now I'm not entirely sure if we're tracking all test issues here, including manual ones

SebastianCzapla, 2025-04-17:
Functionality of VBO001 - VBO003 has been spread around various parts of OSFV, and as of today, these tests are not even part of manual testing. Closing this issue, as it is no longer relevant.

closed 2025-04-17 by SebastianCzapla