Dasharo/open-source-firmware-validation #359: Create a test suite which verifies proper platform setup in the lab

macpijan, 2024-07-05:
We have something like: https://github.com/Dasharo/open-source-firmware-validation/blob/develop/dasharo-compatibility/basic-interfaces.robot

But in reality, setting up platform in the lab typically involves of more activity. Each of them should be verified by such a suite, so it can be used as a way to automatically verify the task of setting up new platform in the lab. This suite can also be started **before** regression testing to ensure the platform is ready for testing. Nowadays, we often find some setup is missing **during** regression testing. Detecting problems **later** in the release process is counterproductive and costs us more in the end. We should focus on detecting problems as early as possible.

What should be verified **may** be platform dependent, but typically it involves the following:
- [ ] Power control - both relay/sonoff and power/reset buttons if avilable
- [ ] Serial / PiKVM - depending on DUT_CONNECTION_TYPE
- [ ] **external flashing**
- [ ] OSes boot (typically Ubuntu and Windows) - including proper credentials / hostname verification; in some cases for the new platform we do not have Dasharo firmware yet, so this must be done manually still

Further step would be to verify hardware matrix as well, but this is big enough of a task to be tracked separately.

BeataZdunczyk, 2024-08-14:
@JanPrusinowski   isn't this one completed? Please provide a status update with links here.

SebastianCzapla, 2025-04-17:
As of 17.04.2025, util/basic-platform-setup.robot covers functionality from this issue, as well as sets up required dependency. Closing this issue, as it is complete, and quite outdated.

closed 2025-04-17 by SebastianCzapla