Dasharo/open-source-firmware-validation #364: ME Disabled (HAP) will no longer affect lspci output

filipleple, 2024-07-09:
### Device

Novacustom v5x

### RTE version

--

### OSFV version

9603436e11cf52ab699fe4e928cdb6acb106198a

### Affected component(s) or functionality

[ME-neuter test](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/dasharo-security/me-neuter.robot#L66)

### Brief summary

ME Disabled (HAP) will no longer affect lspci output, so we need to verify it in some other way.

### How reproducible

100%

### How to reproduce

Run the [MNE004.001 Intel ME mode option Disabled (HAP) works correctly (Ubuntu)](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/dasharo-security/me-neuter.robot#L66) case

### Expected behavior

It should pass

### Actual behavior

It will always fail, since the HAP disabling doesn't affect `lspci` output anymore - ME will still appear as 00:16.0

### Link to screenshots or logs

--

macpijan, 2024-07-11:
We have a change in progress here: https://github.com/Dasharo/open-source-firmware-validation/pull/346
But it seems it's been applied to static test, can we apply this to dynamic checks as well @wiktormowinski ?
Perhaps we can have a keyword to check ME state, and reuse it across these tests.

wiktormowinski, 2024-07-12:
I just did the same in 3 test cases from me-neuter. By ''the same'' i mean the no-keyword version ... for now of course.
Same place:
https://github.com/Dasharo/open-source-firmware-validation/pull/346

BeataZdunczyk, 2024-07-16:
still working on fix here: https://github.com/Dasharo/open-source-firmware-validation/pull/375

mkopec, 2024-07-16:
If ME is enabled, we can check Operation Mode field in HFSTS1. When the HAP bit is set, the ME should go into Debug Mode, indicated by the Current Operation Mode field set to 2.

Or check CBMEM logs:

```
[DEBUG]  ME: HFSTS1                      : 0x80022054
[DEBUG]  ME: Current Operation Mode      : 2
[INFO ]  HECI: coreboot in recovery mode; found CSE in expected Debug Mode state, skipping EOP
```

wiktormowinski, 2024-08-06:
Newest version working on the new Develop branch https://github.com/Dasharo/open-source-firmware-validation/pull/403

JanPrusinowski, 2024-10-16:
I'm having issue on MSI where disabling ME (both HAP and Soft) wont output correct output in linux when running `setpci -s 16.0 40.L` - this breaks me-neuter tests
[image]

wiktormowinski, 2024-10-16:
maybe on MSI roms Intel ME doesn't get initialized at all if we disable it.
Same thing happens on Z690-A DDR5, ME shows up on lspci only when set to Enabled.

macpijan, 2024-10-18:
I think we already do that here: https://github.com/Dasharo/open-source-firmware-validation/blob/develop/lib/me.robot#L27

macpijan, 2024-10-23:
@BeataZdunczyk @JanPrusinowski Should be relatively easy to improve this test by adding one more branch:
1. If we disable ME, we check if PCI device is present first
  - if it is not, the test can PASS, as the ME is disabled
  - if it is present, we must check the setpci command output as we do now.
  
  This would make it work automatically for next test iteration on MSI.

wiktormowinski, 2024-11-12:
ok the description problem is being resolved here: https://github.com/Dasharo/open-source-firmware-validation/issues/536
while this problem got fixed: https://github.com/Dasharo/open-source-firmware-validation/pull/579