Dasharo/open-source-firmware-validation #369: Verified Boot tests FAIL

philipandag, 2024-07-11:
### Device

VP6670

### RTE version

d45acd2253d9

### OSFV version

f65343f6ee8b

### Affected component(s) or functionality

All of the automated Verified Boot tests

### Brief summary

The automatic tests fail, manual testing works fine 

### How reproducible

Seems to happen every time on VP6670

### How to reproduce

Run verified-boot suite on vp6670, ID00004, for example:
```
robot -L TRACE -v rte_ip:192.168.10.110 -v device_ip:192.168.10.136 -v sonoff_ip:192.168.10.113  -v config:protectli-vp6670 -v fw_file:../protectli_vp66xx_v0.9.0-rc6.rom dasharo-security/verified-boot.robot 
```

### Expected behavior

The tests should PASS because testing the functionality manually works

### Actual behavior

The tests fail:
```
λ robot -L TRACE -v rte_ip:192.168.10.110 -v device_ip:192.168.10.136 -v sonoff_ip:192.168.10.113  -v config:protectli-vp6670 -v snipeit:no -v fw_file:../protectli_vp66xx_v0.9.0-rc6.rom dasharo-security/verified-boot.robot 
==============================================================================
Verified-Boot                                                                 
==============================================================================
VBO006.002 Check whether the verstage was run :: Check whether the... | FAIL |
Parent suite setup failed:
No match found for 'login:' in 5 minutes. Output:
------------------------------------------------------------------------------
```

### Link to screenshots or logs

https://cloud.3mdeb.com/index.php/s/an3nJA7cZZZsciw

JanPrusinowski, 2024-07-12:
Working at: [https://github.com/Dasharo/open-source-firmware-validation/tree/369-vp6670-verified-boot-tests-fail](https://github.com/Dasharo/open-source-firmware-validation/tree/369-vp6670-verified-boot-tests-fail)

BeataZdunczyk, 2024-07-19:
Isn't this fixed, @JanPrusinowski? Please share the platform on which fix was successfully tested, and close this issue.

JanPrusinowski, 2024-07-19:
This issue was resolved on vp6670 platform.

closed 2024-07-19 by JanPrusinowski