Dasharo/open-source-firmware-validation #371: VP6670 tests can't install required software

philipandag, 2024-07-11:
### Device

VP6650

### RTE version

d45acd2253d9

### OSFV version

f65343f6ee8b

### Affected component(s) or functionality

The CPU-core-count test,  the process of downloading `cpuid`

### Brief summary

The test supposedly downloads `cpuid` then fails because it is not found

### How reproducible

Every time if cpuid is not installed prior

### How to reproduce

Run any `CCC` test using cpuid, like:
`robot -L TRACE -v rte_ip:192.168.10.14 -v device_ip:192.168.10.156 
-v sonoff_ip:192.168.10.119  -v config:protectli-vp6650 -t "CCC
003*" dasharo-compatibility/cpu-cores-count.robot`

### Expected behavior

The `cpuid` should be installed and the test should pass

### Actual behavior

The test fails because `cpuid` package is not installed despite the test saying it has installed the package.

### Link to screenshots or logs

Logs: https://cloud.3mdeb.com/index.php/s/RLyC3Sir4Fqtn55
```
==============================================================================
Cpu-Cores-Count                                                               
==============================================================================
CCC003.001 Check core count (HT Enabled, P: All, E: 0) (Ubuntu)       .       
Checking if cpuid is installed...

Package cpuid is not installed

Installing required package (cpuid)...

Required package (cpuid) installed successfully
CCC003.001 Check core count (HT Enabled, P: All, E: 0) (Ubuntu)       | FAIL |
'Command 'cpuid' not found, but can be installed with:
apt install cpuid
0' cannot be converted to an integer: ValueError: invalid literal for int() with base 10: "command'cpuid'notf
ound,butcanbeinstalledwith:aptinstallcpuid0"
------------------------------------------------------------------------------
Cpu-Cores-Count                                                       | FAIL |
Suite teardown failed:
Several failures occurred:

1) No connection open

2) No connection open

3) No connection open

4) No connection open
```

macpijan, 2024-07-11:
Following diff might help here:

```
diff --git a/keywords.robot b/keywords.robot
index 1a8dafef06d5..e5e45c5925e3 100644
--- a/keywords.robot
+++ b/keywords.robot
@@ -1298,7 +1298,7 @@ Detect Or Install Package
     Install Package    ${package}
     Sleep    10s
     ${is_package_installed}=    Check If Package Is Installed    ${package}
-    IF    ${is_package_installed}=='False'
+    IF    not ${is_package_installed}
         FAIL    \nRequired package (${package}) cannot be installed
     END
     Log To Console    \nRequired package (${package}) installed successfully
```

macpijan, 2024-07-11:
On top of that, you had network error in the log:

```
 	Return: "Reading package lists... Done\nBuilding dependency tree... Done\nReading state information... Done\nThe following package was automatically installed and is no longer required:\n  libsysmetrics1\nUse 'sudo apt autoremove' to remove it.\nThe following NEW packages will be installed:\n  cpuid\n0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.\nNeed to get 104 kB of archives.\nAfter this operation, 443 kB of additional disk space will be used.\nIgn:1 http://pl.archive.ubuntu.com/ubuntu jammy/universe amd64 cpuid amd64 20211210-1\nIgn:1 http://pl.archive.ubuntu.com/ubuntu jammy/universe amd64 cpuid amd64 20211210-1\nIgn:1 http://pl.archive.ubuntu.com/ubuntu jammy/universe amd64 cpuid amd64 20211210-1\nErr:1 http://pl.archive.ubuntu.com/ubuntu jammy/universe amd64 cpuid amd64 20211210-1\n  Temporary failure resolving 'pl.archive.ubuntu.com'\nE: Failed to fetch http://pl.archive.ubuntu.com/ubuntu/pool/universe/c/cpuid/cpuid_20211210-1_amd64.deb  Temporary failure resolving 'pl.archive.ubuntu.com'\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?"
```

macpijan, 2024-07-16:
@BeataZdunczyk Proposed some hints already how to resolve this

philipandag, 2024-07-22:
> Following diff might help here:
> 
> ```
> diff --git a/keywords.robot b/keywords.robot
> index 1a8dafef06d5..e5e45c5925e3 100644
> --- a/keywords.robot
> +++ b/keywords.robot
> @@ -1298,7 +1298,7 @@ Detect Or Install Package
>      Install Package    ${package}
>      Sleep    10s
>      ${is_package_installed}=    Check If Package Is Installed    ${package}
> -    IF    ${is_package_installed}=='False'
> +    IF    not ${is_package_installed}
>          FAIL    \nRequired package (${package}) cannot be installed
>      END
>      Log To Console    \nRequired package (${package}) installed successfully
> ```
> 
> The `Check If Package Is Installed` returns bool, not string

That's right, at least now robot realises the package was not installed.

philipandag, 2024-07-22:
> On top of that, you had network error in the log:
>
> ```
>  	Return: "Reading package lists... Done\nBuilding dependency tree... Done\nReading state information... Done\nThe following package was [...]

And that is the issue here. It seems like for some reason when the installation is preformed by robot it fails because of fail in name resolution. Although when I copy&paste the exact command that is sent via serial  the package installs succesfully every time.

macpijan, 2024-07-22:
> Although when I copy&paste the exact command that is sent via serial the package installs succesfully every time.

It is most likely a timing problem, try waiting for network to become available prior downloading packages.

philipandag, 2024-07-22:
More curiously, after a couple attempts it seems that the problem vanished and the packages are being installed succesfully

philipandag, 2024-07-22:
> It is most likely a timing problem, try waiting for network to become available prior downloading packages.
It may be just it. Maybe the device needs a little more time to get the nameservers from dhcp. I will try that.

philipandag, 2024-07-22:
I will run the test a couple times in a loop and see if it is more stable with a little bit of delay before installing the package.

macpijan, 2024-07-22:
> if it is more stable with a little bit of delay before installing the package.

More preferably, add a keyword checking for network connection. Perhaps we already have one.

macpijan, 2024-07-22:
Something like

```
    Wait Until Keyword Succeeds    5x    1s
    ...    Check Internet Connection On Linux
```

philipandag, 2024-07-22:
I have tested installing after reboot then uninstalling the `cpuid` package in loop for 10 times and it did not fail once. Seems it doesn't happen that often, but when it was happening it was consistent for some time. I will try adding your suggestion and running a longer test.

philipandag, 2024-07-22:
Now that is unfortunate
```
STR001.001 Test :: Testing if Detect Or Install Package kw is stable  | FAIL |
Keyword 'Check Internet Connection On Linux' failed after retrying 10 times. The last error was: 'ping: google-public-dns-a.google.com: Temporary failure in name resolution' does not contain ', 0% packet loss'
```
It seems sometimes the device sometimes doesn't login to the network even after over 10s

philipandag, 2024-07-22:
pr: https://github.com/Dasharo/open-source-firmware-validation/pull/400

closed 2024-07-26 by macpijan