Dasharo/open-source-firmware-validation #377: Document (in error message) what happens if telnet conenction is open somewhere else, and whac can we do about that

macpijan, 2024-07-15:
1. Open telnet and try running tests. Note the error messages in tests.
  - it should be something like: `EOFError: telnet connection closed`
2. Locate the place in code where it might happen
3. Add some TRY/EXCEPT and error handling logic, to inform user on the possible problems and resolutions:
  - telnet (or minicom on RTE) connection might be opened - either by you or someone else
  - verify and close your connections before starting tests
  - stop minicom processes on RTE, restart ser2net service (`systemctl restart ser2net`)

wiktormowinski, 2024-07-16:
is this the desired outcome? https://github.com/Dasharo/open-source-firmware-validation/pull/382

m-iwanicki, 2024-07-16:
@macpijan @wiktormowinski Wouldn't the most likely culprit be [Serial Setup](https://github.com/Dasharo/open-source-firmware-validation/blob/7c18adbea823700a5d8a06cd750a5f3b4ea8fca0/keywords.robot#L25) keyword?
It is used in the beginning when preparing test suite. If before starting test another client was connected to telnet then I don't think test would ever go past this point.

macpijan, 2024-07-16:
> Wouldn't the most likely culprit be [Serial Setup](https://github.com/Dasharo/open-source-firmware-validation/blob/7c18adbea823700a5d8a06cd750a5f3b4ea8fca0/keywords.robot#L25) keyword?
It is used in the beginning when preparing test suite. If before starting test another client was connected to telnet then I don't think test would ever go past this point.

Agreed, the problem is almost always exclusively in the Suite Setup. It is very unlikely to happen elsewhere, so if guarding every call somehow globally does not work, let's focus on this critical area.