Dasharo/open-source-firmware-validation #385: VP66xx AUD001.002 Windows doesn't detect sound subsystem

philipandag, 2024-07-17:
### Device

VP6670

### RTE version

protectli_vp66xx_v0.9.0-rc6

### OSFV version

osfv-integration-regression

### Affected component(s) or functionality

Audio subsystem in windows detection AUD001.002

### Brief summary

The AUD001.002 test method does not work

### How reproducible

100%

### How to reproduce

Run the AUD001.002 test automatically or manually by writing the command to the powershell:
```
Get-WmiObject -class Win32_SoundDevice
```

### Expected behavior

Audio devices should be listed

### Actual behavior

there is no output

### Link to screenshots or logs

[image]

### Additional context

Related:
 https://github.com/Dasharo/dasharo-issues/issues/947

JanPrusinowski, 2024-07-29:
Same issue on v1210

macpijan, 2024-08-16:
It should a matter of the proper driver installation procedures, especially since we could make it work on V1210.