Dasharo/open-source-firmware-validation #386: Automate new platform config setup

BeataZdunczyk, 2024-07-17:
### The problem you're addressing (if any)

Not fully prepared platform configs for release test execution.

### Describe the solution you'd like

- [ ] Clear list of platform config needed for full regression tests
    - [ ] Determine and prepare list of configs that need to be set
    - [ ] Prepare test for checking if needed configs are missing
- [ ] Automate extraction of information from HCL reports to provide input for filling platform configs
    - [ ] Determine what information is currently available from HCL reports.
    - [ ] Expand the HCL reports to include as much information as needed for adding new platform into the test environment
    - [ ] If some information is not possible to obtain from HCL reports, then add tips on where to get this information, eg. from processor documentation
    
 Ultimately, aim to automate this process as much as possible.

### Where is the value to a user, and who might that user be?

This benefits developers and testers by automating adding new platform to test environment, making the testing process faster.

### Describe alternatives you've considered

none

### Additional context

Refer to the current flow: [Adding New Platforms](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/docs/adding-new-platforms.md)

macpijan, 2024-07-22:
This mostly sounds like a duplicate of https://github.com/Dasharo/open-source-firmware-validation/issues/386 to me.
Maybe a bit more defined into steps.

From my perspective, we have done a first step with generating variables for configs.

Now, we should proceed with config validation, to detect problems with missing variables sooner than during regression executino.

BeataZdunczyk, 2024-07-23:
Okay, we are abandoning the stronger use of HCL reports, I'm closing this issue

closed 2024-07-23 by BeataZdunczyk