Dasharo/open-source-firmware-validation #401: Make boot time measure work on qemu

philipandag, 2024-07-22:
### The problem you're addressing (if any)

dasharo-performance/boot-time-measure.robot is not prepared to run on qemu

### Describe the solution you'd like

modify the test so that it works on qemu

### Where is the value to a user, and who might that user be?

us, running boot-time-measure on qemu

### Describe alternatives you've considered

running it on other devices instead - it is slower and requires a free platform

### Additional context

Running on qemu with ubuntu 22.04 lts:
```
./qemu-run.sh graphic os
robot  -L TRACE -v config:qemu -v rte_ip:127.0.0.1 -v snipeit:no dasharo-performance/boot-time-measure.robot
```

Running it on qemu results in an error probably related to the test trying to use RTE
```
==============================================================================
Boot-Time-Measure                                                             
==============================================================================
CBMEM001.001 Serial boot time measure: coreboot booting time after... ..

CBMEM001.001 Serial boot time measure: coreboot booting time after... | FAIL |
Non-existing index or alias 'RteCtrl'.
------------------------------------------------------------------------------
CBMEM002.001 Serial boot time measure: coreboot booting time after... .

CBMEM002.001 Serial boot time measure: coreboot booting time after... | FAIL |
Unknown connection method:
------------------------------------------------------------------------------
CBMEM003.001 Serial boot time measure: coreboot booting time after... | FAIL |
Unknown connection method:
------------------------------------------------------------------------------
Boot-Time-Measure                                                     | FAIL |
3 tests, 0 passed, 3 failed
==============================================================================
```

Logs:
https://cloud.3mdeb.com/index.php/s/5tHkpwtLCx8oxXs

mkopec, 2024-08-09:
resolved by https://github.com/Dasharo/open-source-firmware-validation/pull/399 @philipandag  ?

philipandag, 2024-08-12:
> resolved by https://github.com/Dasharo/open-source-firmware-validation/pull/399 @philipandag ?
@mkopec I don't think so, that pr only added richer measurements of the boot time. This error seems to be related to osfv configs, e.g.
```
CBMEM003.001 Serial boot time measure: coreboot booting time after... | FAIL |
Unknown connection method:
```
`${DUT_CONNECTION_METHOD}` seems to not be set there.

PLangowski, 2024-08-13:
I managed to fixed the error in https://github.com/Dasharo/open-source-firmware-validation/pull/456, however I'm getting the following error: 
```
Could not open /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq: No such file or directory
```
@philipandag Could you check if you also get that error?

PLangowski, 2024-08-14:
What if you run it with
```
robot -L TRACE -v config:qemu -v rte_ip:127.0.0.1 -v snipeit:no dasharo-performance/boot-time-measure.robot
```

PLangowski, 2024-08-14:
@mkopec Do you know how the following error can be resolved?

```
Could not open /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq: No such file or directory
```

philipandag, 2024-08-14:
> What if you run it with

Doesn't make a difference. I have tried to create a clean osfv repo testing environment but it didn't help.

PLangowski, 2024-08-21:
@philipandag could you send the full log here?

PLangowski, 2024-08-23:
I'm not sure how to enable `cpufreq` on QEMU. The Ubuntu kernel config seems to support it

```
CONFIG_CPU_FREQ=y
CONFIG_CPU_FREQ_GOV_ATTR_SET=y
CONFIG_CPU_FREQ_GOV_COMMON=y
CONFIG_CPU_FREQ_STAT=y
# CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_USERSPACE is not set
CONFIG_CPU_FREQ_DEFAULT_GOV_SCHEDUTIL=y
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
CONFIG_CPU_FREQ_GOV_POWERSAVE=y
CONFIG_CPU_FREQ_GOV_USERSPACE=y
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
CONFIG_CPU_FREQ_GOV_CONSERVATIVE=y
CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y

CONFIG_X86_ACPI_CPUFREQ=y
```

but even after executing `modprobe acpi-cpufreq` I'm encountering the same issue. This is probably a problem with the emulated cpu

```
# lscpu

Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         40 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  2
  On-line CPU(s) list:   0,1
Vendor ID:               GenuineIntel
  Model name:            QEMU Virtual CPU version 2.5+
    CPU family:          15
    Model:               107
    Thread(s) per core:  1
    Core(s) per socket:  2
    Socket(s):           1
    Stepping:            1
    BogoMIPS:            6385.49
    Flags:               fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cm
                         ov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx lm
                          constant_tsc nopl xtopology cpuid tsc_known_freq pni c
                         x16 x2apic hypervisor lahf_lm cpuid_fault pti
Virtualization features:
  Hypervisor vendor:     KVM
  Virtualization type:   full
Caches (sum of all):
  L1d:                   64 KiB (2 instances)
  L1i:                   64 KiB (2 instances)
  L2:                    8 MiB (2 instances)
  L3:                    16 MiB (1 instance)
NUMA:
  NUMA node(s):          1
  NUMA node0 CPU(s):     0,1
Vulnerabilities:
  Gather data sampling:  Not affected
  Itlb multihit:         KVM: Mitigation: VMX unsupported
  L1tf:                  Mitigation; PTE Inversion
  Mds:                   Vulnerable: Clear CPU buffers attempted, no microcode;
                         SMT Host state unknown
  Meltdown:              Mitigation; PTI
  Mmio stale data:       Unknown: No mitigations
  Retbleed:              Not affected
  Spec rstack overflow:  Not affected
  Spec store bypass:     Vulnerable
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer
                          sanitization
  Spectre v2:            Mitigation; Retpolines; STIBP disabled; RSB filling; PB
                         RSB-eIBRS Not affected; BHI Retpoline
  Srbds:                 Not affected
  Tsx async abort:       Not affected
```