Dasharo/open-source-firmware-validation #412: minimal-regression not working

miczyg1, 2024-07-26:
### Device

Minnowbaord Turbot

### RTE version

v0.5.3

### OSFV version

develop (1a5e1bdd3bdc34427c8e0c114eab13de5fc0ccb3)

### Affected component(s) or functionality

minimal-regression

### Brief summary

minimal-regression throws errors

### How reproducible

Always

### How to reproduce

README says that `$TEST_SUITE` variable can be a `uefi-shell.robot` but it clearly doesn't work with minimal-regression:

```
DEVICE_IP=192.168.10.60 RTE_IP=192.168.10.116 CONFIG=minnowboard-turbot ./scripts/run.sh dasharo-compatibility/uefi-shell.robot -- --include "minimal-regression"
robot -L TRACE -l logs/minnowboard-turbot/2024_07_26_12_18_41/uefi-shell.robot_log.html -r logs/minnowboard-turbot/2024_07_26_12_18_41/uefi-shell.robot_report.html -o logs/minnowboard-turbot/2024_07_26_12_18_41/uefi-shell.robot_out.xml -b logs/minnowboard-turbot/2024_07_26_12_18_41/uefi-shell.robot_debug.log -v rte_ip:192.168.10.116 -v config:minnowboard-turbot -v device_ip:192.168.10.60 --include minimal-regression dasharo-compatibility/uefi-shell.robot
[ ERROR ] Suite 'Uefi-Shell' contains no tests matching tag 'minimal-regression'.

Try --help for usage information.
```

I think it should be `$TEST_MODULE`?

Then we have a complaint on missing FW_FILE, which also isn;t documented in README:

```
DEVICE_IP=192.168.10.60 RTE_IP=192.168.10.116 CONFIG=minnowboard-turbot ./scripts/run.sh dasharo-compatibility -- --include "minimal-regression"
robot -L TRACE -l logs/minnowboard-turbot/2024_07_26_12_20_27/dasharo-compatibility_log.html -r logs/minnowboard-turbot/2024_07_26_12_20_27/dasharo-compatibility_report.html -o logs/minnowboard-turbot/2024_07_26_12_20_27/dasharo-compatibility_out.xml -b logs/minnowboard-turbot/2024_07_26_12_20_27/dasharo-compatibility_debug.log -v rte_ip:192.168.10.116 -v config:minnowboard-turbot -v device_ip:192.168.10.60 --include minimal-regression dasharo-compatibility
==============================================================================
Dasharo-Compatibility                                                         
==============================================================================
Dasharo-Compatibility.Auto-Boot-Time-Out                                      
==============================================================================
BMM002.001 F9 resets Auto Boot Time-out to default value :: Check ... | FAIL |
Parent suite setup failed:
Variable '${FW_FILE}' does not exist.
```

If we pass a valid FW_FILE, the new have another error:

```
FW_FILE=/home/miczyg/Projects/Dasharo/coreboot/intel_minnowmax_v0.9.0-rc1.rom DEVICE_IP=192.168.10.60 RTE_IP=192.168.10.116 CONFIG=minnowboard-turbot ./scripts/run.sh dasharo-compatibility -- --include "minimal-regression"
robot -L TRACE -l logs/minnowboard-turbot/2024_07_26_12_21_07/dasharo-compatibility_log.html -r logs/minnowboard-turbot/2024_07_26_12_21_07/dasharo-compatibility_report.html -o logs/minnowboard-turbot/2024_07_26_12_21_07/dasharo-compatibility_out.xml -b logs/minnowboard-turbot/2024_07_26_12_21_07/dasharo-compatibility_debug.log -v rte_ip:192.168.10.116 -v config:minnowboard-turbot -v device_ip:192.168.10.60 -v fw_file:/home/miczyg/Projects/Dasharo/coreboot/intel_minnowmax_v0.9.0-rc1.rom --include minimal-regression dasharo-compatibility
==============================================================================
Dasharo-Compatibility                                                         
==============================================================================
Dasharo-Compatibility.Auto-Boot-Time-Out                                      
==============================================================================
BMM002.001 F9 resets Auto Boot Time-out to default value :: Check ... | FAIL |
Parent suite setup failed:
Initializing library 'osfv.rf.rte_robot.RobotRTE' with arguments [ 192.168.10.116 | True ] failed: Failed to retrieve model name from Snipe-IT. Check again arguments, or try providing model manually.
Traceback (most recent call last):
  File "/home/miczyg/Projects/Dasharo/open-source-firmware-validation/venv/lib/python3.12/site-packages/osfv/rf/rte_robot.py", line 42, in __init__
    raise AssertionError(
AssertionError: Failed to retrieve model name from Snipe-IT. Check again arguments, or try providing model manually.
```

At this point I gave up...

### Expected behavior

README is giving good instructions how to run minimal-regression

The minimal-regression is supposed to work on boards supported by OSFV

### Actual behavior

README is inaccurate how to run minimal-regression

The minimal-regression throws errors

### Link to screenshots or logs

Logs in the reproducibility section

macpijan, 2024-07-26:
> README says that $TEST_SUITE variable can be a uefi-shell.robot but it clearly doesn't work with minimal-regression:

It can and it works. It simply says that in this suite does not have such tags. Not each suite will have such tags.

> Then we have a complaint on missing FW_FILE, which also isn;t documented in README:

It is in several places, especially the one on running regression.
Some suites need firmware for flashing, some not.

> If we pass a valid FW_FILE, then we have another error:

Either in your command, or in snipeit we have an error.
In snipeit the RTE_IP is 192.168.10.112. Which is the correct one?

macpijan, 2024-07-26:
@philipandag Based on feedback from this issue, can you try to improve readme? My conrete suggesions:

- [ ] provide a better example here: https://github.com/Dasharo/open-source-firmware-validation/tree/develop?tab=readme-ov-file#running-tests-with-additional-arguments - maybe add another robot argument, as test case name?
- [ ] Document the `minimal regression` to work with `regression.sh` under this section: https://github.com/Dasharo/open-source-firmware-validation/tree/develop?tab=readme-ov-file#running-regression-tests

miczyg1, 2024-07-26:
I have refreshed the osfv_cli and reinstalled it, however getting an error when executing make i nthe osfv-scripts/osfv_cli directory:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pyopenssl 24.1.0 requires cryptography<43,>=41.0.5, but you have cryptography 43.0.0 which is incompatible.
```

With the newer osfv_cli I got it working.

philipandag, 2024-07-26:
I think i added an example of using the additional args in `regression.sh` but it must have got lost during rebasing :(

philipandag, 2024-08-06:
@miczyg1 do you consider the issue solved? Please close it if so.

closed 2024-08-06 by miczyg1