Dasharo/open-source-firmware-validation #414: Protectli v1210 cnat dettect keyboard

JanPrusinowski, 2024-07-29:
### Device

Protectli v1210

### RTE version

v0.9.3-rc1

### OSFV version

develop

### Affected component(s) or functionality

DUT can't detect the keyboard.

### Brief summary

DUT can't detect the keyboard.

### How reproducible

100%

### How to reproduce

robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.55 -v device_ip:192.168.10.252 -v config:protectli-v1210 dasharo-compatibility/usb-hid-and-msc-support.robot

### Expected behavior

Tests pass

### Actual behavior

Tests fail

### Link to screenshots or logs

janprusinowski in ~/Desktop/Dasharo2/open-source-firmware-validation on protectli_v1210_0.9.3-rc1-testing ● ● λ robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.55 -v device_ip:192.168.10.252 -v config:protectli-v1210 -t 'USB002.003*' dasharo-compatibility/usb-hid-and-msc-support.robot
==============================================================================
Usb-Hid-And-Msc-Support                                                       
==============================================================================
USB002.003 USB keyboard in OS (Windows) :: Check whether the exter... | FAIL |
'' should not be empty.
------------------------------------------------------------------------------
Usb-Hid-And-Msc-Support                                               | FAIL |
1 test, 0 passed, 1 failed
==============================================================================

### Additional context

Most probably there is no keyboard of anykind connected to the DUT

wiktormowinski, 2024-07-30:
The exact same problem occurs on: 
 - Protectli v1410 vault_jsl
   FW Version: v0.9.3-rc1

JanPrusinowski, 2024-08-05:
same issue on v1211