Dasharo/open-source-firmware-validation #415: DCU tests fail on Protectli v1210

JanPrusinowski, 2024-07-29:
### Device

Protectli v1210

### RTE version

v0.9.3-rc1

### OSFV version

develop

### Affected component(s) or functionality

DCU001-3 tests

### Brief summary

Those tests fail to start

### How reproducible

100%

### How to reproduce

```robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.55 -v device_ip:192.168.10.252 -v config:protectli-v1210 -v fw_file:/home/janprusinowski/Downloads/protectli_v1210_v0.9.3-rc1.rom dasharo-compatibility/dcu.robot
```

### Expected behavior

Thests start and run

### Actual behavior

Tests fail to start:
```
==============================================================================
Dcu                                                                           
==============================================================================
DCU001.001 Change the UUID :: This test case verifies that the UUI... | FAIL |
'' does not contain 'Success'
------------------------------------------------------------------------------
DCU002.001 Change the serial number :: This test case verifies tha... | FAIL |
'' does not contain 'Success'
------------------------------------------------------------------------------
DCU003.001 Change the bootsplash logo :: This test case verifies t... | FAIL |
'' does not contain 'Success'
------------------------------------------------------------------------------
DCU004.001 Verify SMMSTORE changes :: This test case verifies that... | SKIP |
smmstoretool ins not yet supported in DCU
------------------------------------------------------------------------------
Dcu                                                                   | FAIL |
4 tests, 0 passed, 3 failed, 1 skipped
==============================================================================
Output:  /home/janprusinowski/Desktop/Dasharo2/open-source-firmware-validation/output.xml
Log:     /home/janprusinowski/Desktop/Dasharo2/open-source-firmware-validation/log.html
Report:  /home/janprusinowski/Desktop/Dasharo2/open-source-firmware-validation/report.html
```

### Link to screenshots or logs

[log.zip](https://github.com/user-attachments/files/16411304/log.zip)

wiktormowinski, 2024-07-31:
The exact same problem occurs on:
    - Protectli v1410 vault_jsl
      FW Version: v0.9.3-rc1

    - Protectli v1610 vault_jsl
      FW Version: v0.9.3-rc1

SebastianCzapla, 2025-04-25:
Repro steps are missing, but the fix was merged.

It seems like an older, known issue, note about it added here: https://github.com/Dasharo/dcu/pull/20

closed 2025-04-25 by SebastianCzapla