Dasharo/open-source-firmware-validation #426: TPM Support test only considers TPM2

filipleple, 2024-08-02:
### Device

any

### RTE version

any

### OSFV version

1a5e1bdd3bdc34427c8e0c114eab13de5fc0ccb3

### Affected component(s) or functionality

[TPM Support test](https://github.com/Dasharo/open-source-firmware-validation/blob/main/dasharo-security/tpm-support.robot)

### Brief summary

TPM Support test only considers TPM2

### How reproducible

100%

### How to reproduce

Run TPM Support test on a platform equipped only with a TPM 1.2

### Expected behavior

The test should verify whether the DUT is equipped with a (any) TPM and is able to communicate with it. It probably should first check the version of the platform's TPM and proceed accordingly.

### Actual behavior

The test only considers a TPM2, using `tpm2_tools` and `Should Contain    ${out}    TPM2 log`. 

### Link to screenshots or logs

```
==============================================================================
Tpm-Support
==============================================================================
TPM001.001 TPM Support (firmware) :: This test aims to verify that... | FAIL |
Keyword 'Check Internet Connection On Linux' failed after retrying 5 times. The last error was: 'ping: g
oogle-public-dns-a.google.com: Temporary failure in name resolution' does not contain ', 0% packet loss'
------------------------------------------------------------------------------
TPM001.002 TPM Support (Ubuntu) :: Check whether the TPM is initia... .....
Checking if tpm2-tools is installed...

Package tpm2-tools is installed
TPM001.002 TPM Support (Ubuntu) :: Check whether the TPM is initia... | FAIL |
'ERROR:tcti:src/tss2-tcti/tcti-device.c:452:Tss2_Tcti_Device_Init() Failed to open specified TCTI device
 file /dev/tpmrm0: No such file or directory
ERROR:tcti:src/tss2-tcti/tctildr-dl.c:154:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-
device.so.0
ERROR:tcti:src/tss2-tcti/tcti-device.c:503:Tss2_Tcti_Device_Init() timeout waiting for response from fd 3
ERROR:sys:src/tss2-sys/api/Tss2_Sys_Execute.c:113:Tss2_Sys_ExecuteFinish() Unsupported device. The device is a TPM 1.2
ERROR:esys:src/tss2-esys/api/Esys_GetCapability.c:307:Esys_GetCapability_Finish() Received a non-TPM Error
ERROR:esys:src/tss2-esys/api/Esys_GetCapability.c:107:Esys_GetCapability() Esys Finish ErrorCode (0x00080001)
ERROR: Esys_GetCapability(0x80001) - sys:Catch all for all errors not otherwise specified
ERROR: Unable to run tpm2_pcrread' does not contain 'sha1:'
------------------------------------------------------------------------------
TPM001.003 TPM Support (Windows) :: Check whether the TPM is initi... | SKIP |
TPM001.003 not supported
------------------------------------------------------------------------------
TPM002.001 Verify TPM version (firmware) :: This test aims to veri... | FAIL |
Keyword 'Check Internet Connection On Linux' failed after retrying 5 times. The last error was: 'ping: google-public-dns-a.google.com: Temporary failure in name resolution' does not contain ', 0% packet loss'
------------------------------------------------------------------------------
TPM002.002 Verify TPM version (Ubuntu) :: This test aims to verify... | FAIL |
'1' does not contain '2'
------------------------------------------------------------------------------
TPM002.003 Verify TPM version (Windows) :: This test aims to verif... | SKIP |
TPM002.003 not supported
------------------------------------------------------------------------------
TPM003.001 Check TPM Physical Presence Interface (firmware) :: Thi... | FAIL |
Keyword 'Check Internet Connection On Linux' failed after retrying 5 times. The last error was: 'ping: google-public-dns-a.google.com: Temporary failure in name resolution' does not contain ', 0% packet loss'
------------------------------------------------------------------------------
TPM003.002 Check TPM Physical Presence Interface (Ubuntu) :: This ... | FAIL |
No match found for 'to boot directly' in 30 seconds. Output:
------------------------------------------------------------------------------
TPM003.003 Check TPM Physical Presence Interface (Windows) :: This... | SKIP |
TPM003.003 not supported
------------------------------------------------------------------------------
Tpm-Support                                                           | FAIL |
9 tests, 0 passed, 6 failed, 3 skipped
==============================================================================
```

BeataZdunczyk, 2024-09-05:
@SebastianCzapla we need a link to the PR with the fixes for this issue, along with a status update. Please provide the test results and specify the DUT.

SebastianCzapla, 2024-10-15:
Introduced with https://github.com/Dasharo/open-source-firmware-validation/pull/487

closed 2024-10-15 by SebastianCzapla