Dasharo/open-source-firmware-validation #431: V5xTNx laptops need a powernap after executing `efibootmgr` in CNB001.001

philipandag, 2024-08-05:
### Device

V540TND, V560TNE

### RTE version

v0.9.1-rc1

### OSFV version

fix-novacustom-v5xtnx-configs-powerctrl

### Affected component(s) or functionality

CNB001.001 test

### Brief summary

The test fails and then the device is bricked for about 30 minutes

### How reproducible

seems to appear every time

### How to reproduce

Run CNB001.001

### Expected behavior

The test should pass and the device should not become bricked

### Actual behavior

The device enters a weir state where it seems it is bricked. The keyboard and control leds work fine but it never boots. Pressing power button causes the device to power off immediately suggesting it did not leave firmware

### Link to screenshots or logs

V54 log: https://cloud.3mdeb.com/index.php/s/ba2tz85pH3XyTEC
V56 log: https://cloud.3mdeb.com/index.php/s/TRYzeBPXc3LEXcd

### Additional context

It is hard to reproduce because the laptops become unusable for about 30 minutes after the test. Both eventually managed to boot and work normally after performing multiple cmos resets and then leaving them on for about 30 minutes.

### Solutions you've tried

Still haven't tried doing anything about it. Just finding out if it is reproducible takes a long time.

philipandag, 2024-08-06:
@macpijan that is an interesting behavior that you might want to know about

mkopec, 2024-08-20:
The cause has been narrowed down to commit https://github.com/Dasharo/open-source-firmware-validation/pull/455/commits/a539822bb98268ffd552f1dcc83709fd5d15d0f4 , not sure how to fix it correctly yet

philipandag, 2024-08-20:
@mkopec This commit a539822 does not appear on the branch on which I had the issue. Are you sure it is the cause of the problems?

philipandag, 2024-09-12:
I have been running tests on V540TND on v0.9.1-rc5 for half a day yesterday and the bricks doesn't seem to happen now, even during aggressive rebooting and stress testing. I will report if any bricks happen again.

closed 2024-09-12 by BeataZdunczyk