Dasharo/open-source-firmware-validation #453: Minnowboard cpu temperature measuring problem

wiktormowinski, 2024-08-12:
### Device

MinnowBoard Turbot B41

### RTE version

-

### OSFV version

branch: minnow-update

### Affected component(s) or functionality

temperature sensor

### Brief summary

dasharo-performance/cdpu-temperature.robot tests cannot read temperature

### How reproducible

always

### How to reproduce

run the CPT001.001 or CPT002.001 test cases

### Expected behavior

The temperature should be read and compared with control value

### Actual behavior

instead the output list with temperature values appears to be empty

### Link to screenshots or logs

[cpu-temperature.zip](https://github.com/user-attachments/files/16582583/cpu-temperature.zip)

wiktormowinski, 2024-08-13:
note that `sensors` works just fine :
```
ubuntu@3mdeb:~$ sensors
soc_dts1-virtual-0
Adapter: Virtual device
temp1:        +48.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +49.0°C  (high = +127.0°C, crit = +127.0°C)
Core 1:       +50.0°C  (high = +127.0°C, crit = +127.0°C)
Core 2:       +52.0°C  (high = +127.0°C, crit = +127.0°C)
Core 3:       +52.0°C  (high = +127.0°C, crit = +127.0°C)

soc_dts0-virtual-0
Adapter: Virtual device
temp1:        +49.0°C  

```

miczyg1, 2024-08-14:
@wiktormowinski as you can see there is no "Package id:" string in the output which the test depends on. Fix the test please

wiktormowinski, 2024-08-14:
Oh ok I looked at other platforms, now i get it. 
I didn't realize how close to the solution i actually was with that `sensors` command's output.

works fine now

closed 2024-08-14 by wiktormowinski

BeataZdunczyk, 2024-08-19:
@wiktormowinski Was the test fixed as this PR is now closed? Please link the PR to this issue.

wiktormowinski, 2024-08-20:
https://github.com/Dasharo/open-source-firmware-validation/commit/c922a9d1a9d7d0d41c3ac545846164d969ba2337 that's all, just one word

miczyg1, 2024-09-05:
> [c922a9d](https://github.com/Dasharo/open-source-firmware-validation/commit/c922a9d1a9d7d0d41c3ac545846164d969ba2337) that's all, just one word

Yes, that's all it takes to fix the test for one platform and break it for all the others...

I won't repeat myself so linking the comment: https://github.com/Dasharo/open-source-firmware-validation/pull/441#discussion_r1744988747