Dasharo/open-source-firmware-validation #472: V560TNE v0.9.1-rc3 touchpad on/off function key not working

philipandag, 2024-08-21:
### Device

V560TNE

### RTE version

--

### OSFV version

develop

### Affected component(s) or functionality

turning touchpad on/off using function key

### Brief summary

touchpad on/off function key it not working

### How reproducible

100%

### How to reproduce

press fn+F1

### Expected behavior

should turn touchpad on/off

### Actual behavior

the touchpad works

### Link to screenshots or logs

manual visual test, no logs

### Solutions you've tried

tried using `libinput debug-events --show-keycodes` but is does not show any event

mkopec, 2024-08-21:
ah, we need to add a quirk to udev for this model too

philipandag, 2024-08-22:
This works fine on Windows

closed 2024-08-29 by philipandag