Dasharo/open-source-firmware-validation #474: SUSP test suite fails to connect via ssh

philipandag, 2024-08-21:
### Device

V560TNE, v0.9.1-rc3

### RTE version

--

### OSFV version

develop

### Affected component(s) or functionality

SUSP test suite

### Brief summary

SUSP test suite tests fail during SSHLibrary.Read Until

### How reproducible

100%

### How to reproduce

run 
`scripts/run.sh dasharo-compatibility/platform-suspend-and-resume.robot`

### Expected behavior

Tests should run and pass

### Actual behavior

test fail because "No open connection"

### Link to screenshots or logs

[Uploading platform-suspend-and-resume.robot_log.zip…]()

philipandag, 2024-08-23:
This issue is fixed by adding a missing config variable https://github.com/Dasharo/open-source-firmware-validation/pull/477/commits/ed3b043e55dd64fe568900cd50f4164832c297f8#diff-c6fb4bfec015f4ff07f92a24ac8afa3ceeaf2698a806221148c9aa56f82c1246R26

BeataZdunczyk, 2024-09-05:
@philipandag if the issue is fixed, you can close your issue yourself. ;)

closed 2024-09-05 by BeataZdunczyk