Dasharo/open-source-firmware-validation #480: CPF stress-test-windows script issues

philipandag, 2024-08-27:
### Device

V560TND

### RTE version

--

### OSFV version

v560tne-v0.9.1-rc3-testing-fixes

### Affected component(s) or functionality

CPF cpu frequency tests under load on Windows

### Brief summary

The script is not run automatically, when run manually only uses 4 cores and ends early which is hardcoded

### How reproducible

100% on this device and branch

### How to reproduce

Run a test like CPF-004.004, check the CPU usage, check che stress-test-windows.ps1 script code

### Expected behavior

The cpu should run at ~100% during the whole test

### Actual behavior

The cpu is running at ~1% usage, running the script does stress the 4 hardcoded cores which results in ~25% usage on this device. Even after increasing the hardcoded core count the cpu usage rises to at most ~60%. Moreover the test ends after about a minute but measurements which in this case are performed over the span of 60 minutes.

### Link to screenshots or logs

```
$numCores = 4 # <-- Only 4 cores, should not be hardcoded but determined on per-device basis

foreach ($loopnumber in 1..$numCores){
    Start-Job -ScriptBlock{
    $result = 1
        foreach ($number in 1..40000000){   # <-- Ends after 40mln iterations which takes about a minute, the length of the test
            $result = $result * $number           # should not depend on how fast is the DUT
        }
    }
}

Wait-Job *
Clear-Host
Receive-Job *
Remove-Job *
```

### Additional context

Maybe there is some reliable tool we could use instead of writing a stress script

### Solutions you've tried

Increasing number of workers and iterations, running the script manually

philipandag, 2024-09-17:
I have tried to make the jobs time-bound and to get the core count dynamically:
```
$numCores = (Get-CimInstance Win32_ComputerSystem).NumberOfLogicalProcessors

foreach ($loopnumber in 1..$numCores){
    $end_time = (Get-Date).AddMinutes(60)
    Start-Job -ScriptBlock{
        $result = 7
        while ((Get-Date) -lt ($end_time)){ # Parametrize stress length
            # Is this really the best operation to stress the system?
            $result = $result * $result
        }
    }
}

Wait-Job *
Clear-Host
Receive-Job *
Remove-Job *
```

This script does not work as well. The core count is ok but the jobs still stop after a couple seconds.

philipandag, 2024-10-14:
I forgot to mention it when I encountered it a couple weeks ago, but tests under load on Ubuntu were not stable too. After just a couple minutes the clocks were starting to fall to the 200-400MHz ranges and tools like `htop` were not reporting CPU usage despite specifying that the stress should take exactly 60 minutes in the `stress-ng` command. This might also be worth looking into

philipanda, 2025-08-07:
@macpijan @BeataZdunczyk that is still relevant to this day. Automatic tests don't stress the CPU at all