Dasharo/open-source-firmware-validation #490: Execute UEFI Shell Command gathers only 1 sec of output

JanPrusinowski, 2024-09-06:
### Device

All

### RTE version

All

### OSFV version

Develop

### Affected component(s) or functionality

OSFV Tests

### Brief summary

Execute UEFI Shell Command gathers only 1 sec of output

### How reproducible

100%

### How to reproduce

Use `Execute UEFI Shell Command` and run with a command that outputs logs for longer than 1 sec.

Good example is to run `reset' command and wait for DUT to reboot and wait for prompt.

### Expected behavior

Use `Execute UEFI Shell Command` and get full output up untill next prompt is displayed.

For example after `reboot` command the output should not be empty.

### Actual behavior

Gathers only 1 second of the output.

### Link to screenshots or logs

[image]

In this scenario output is empty, but it should contain everything up until prompt is displayed again.

### Additional context

There is also an issue where `Set DUT Response Timeout` doesn't affect this command. This might need looking into.

SergiiDmytruk, 2024-09-06:
For extra context, this is the current implementation on `develop`
https://github.com/Dasharo/open-source-firmware-validation/blob/e306abc385cda02aeeb638e923ffbcce432e4a63/lib/terminal.robot#L164-L175
with hard-coded `Sleep 1s` and `Read From Terminal` instead of `Read Until Prompt`.

philipandag, 2025-01-21:
The changes from https://github.com/Dasharo/open-source-firmware-validation/pull/528 work fine. It was the hard-coded `1s` wait time and reading without waiting for the prompt that caused the issue.

I've had to fix the capsule updates test suite a bit to use it for testing the changes.

The few errors are caused by the `Execute Manual Step` kwd not working due to missing tkinter on my computer
[capsule-update.robot_log.zip](https://github.com/user-attachments/files/18485763/capsule-update.robot_log.zip)

mkopec, 2025-01-21:
Thanks for confirming, merged and closing as fixed by https://github.com/Dasharo/open-source-firmware-validation/pull/528 .

closed 2025-01-21 by mkopec