Dasharo/open-source-firmware-validation #496: Protectli V1211 sometimes require long time to post

JanPrusinowski, 2024-09-10:
### Device

Protevtli v1211

### RTE version

v0.9.3-rc1

### OSFV version

develop

### Brief summary

Protectli V1211 long time to post

### How reproducible

50-70%

### How to reproduce

1. flash: protectli_v1211_v0.9.3-rc1.rom
2. open a serial connection to the DUT
3. turn the DUT on
4. press 'del' to enter coreboot
5. pick `Reset`

This is when sometimes the post will take 1 sec and sometimes over a minute. If post will be imidiate repeat steps 4-5 a few times.

### Expected behavior

Post should take similar amount of time each time.

### Actual behavior

Sometimes post takes a long time.

### Link to screenshots or logs

[image]

### Solutions you've tried

Flashing fresh rom did not change anything

JanPrusinowski, 2024-09-10:
there is an issue where sometimes after hitting `DEL` to enter BIOS machine takes a long time to display bios (couple of minutes)

JanPrusinowski, 2024-09-11:
To provoke the error you can run:
```
Reset System Stress Test
    Power On
    FOR    ${i}    IN RANGE    50
        Log To Console    Run: ${i}        
        ${sb_menu}=    Enter Secure Boot Menu And Return Construction
        Tianocore Reset System
    END
```

BeataZdunczyk, 2024-09-11:
V1210 is JSL platform, changing the label

dawidlesiak, 2024-09-25:
```
 V1211                                                                          
 Intel(R) Celeron(R) N5105 @ 2.00GHz                 2.00 GHz                   
 Dasharo (coreboot+UEFI) v0.9.3-rc1                  8192 MB RAM @ 2933 MHz     
```
I wasn’t able to reproduce the issue. Tried restarting around 40 times.

closed 2024-10-08 by JanPrusinowski