Dasharo/open-source-firmware-validation #502: Perform Warmboot Using Rtcwake will fail to boot if shutdown took too long

philipandag, 2024-09-13:
### Device

V540TND

### RTE version

--

### OSFV version

develop

### Affected component(s) or functionality

All tests performing warmboots automatically

### Brief summary

RTCwake turns on the device after a given delay. If the device was not turned off when the time passed it won;t turn on.

### How reproducible

On V540TND ~10%. If the shutdown takes longer than the time after which rtcwake is supposed to trigger then in 100% cases the device won't turn back on.

### How to reproduce

Run cases like SUD002, SMW002, NET002, DDET003

### Expected behavior

The device should turn back on after being turned off

### Actual behavior

It won't if the shutdown took too long

### Link to screenshots or logs

In this log you can see that the first iteration of warmbooting took over 5 minutes to login to linux. That's because I have turned on the device manually by pressing the power button. If i would leave it to be it would never boot back and this test suite would fail. When running a full regression such behavior could ruin the whole automatic regression.

[m2-wifi.robot_log.zip](https://github.com/user-attachments/files/16991578/m2-wifi.robot_log.zip)

### Additional context

It might be easily solved by increasing the time passed to rtcwake significantly but that still would not guarantee it would work properly, the shutdown could always just take longer than expected. Moreover the time to perform this test would also be increases significantly in that case. Maybe there is a more reliable way to perform warmboot?

mkopec, 2024-09-13:
I guess we should just increase the rtcwake timer?

philipandag, 2024-09-20:
https://github.com/Dasharo/open-source-firmware-validation/compare/develop...increase-rtcwake%3Dwarmboot-wait-time?expand=1

mkopec, 2024-09-23:
Timer increased in https://github.com/Dasharo/open-source-firmware-validation/pull/512

Closing, let's re-open this if this problem comes up again

closed 2024-09-23 by mkopec