Dasharo/open-source-firmware-validation #503: UTC automatic tests fail on V540TND on v0.9.1-rc5 with UMD05 Pro Rev.e

philipandag, 2024-09-16:
### Device

V540TND

### RTE version

--

### OSFV version

develop -> v540tnd-v0.9.1-rc5-fixes

### Affected component(s) or functionality

UTC automatic tests

### Brief summary

The tests fail because the detected device names are different than the tests require

### How reproducible

100%

### How to reproduce

`scripts/run.sh dasharo-compatibility/usb-type-c.robot dasharo-compatibility/usb-type-c-semiauto.robot`

### Expected behavior

Tests should pass

### Actual behavior

They fail because the device names are different. Manual tests pass. I think the device names are just different but otherwise it works fine.

### Link to screenshots or logs

Tests were forcefully stopped when they reached the semi-auto tests which are not included in the regression testing I'm performing.

[usb-type-c-semiauto.robot_log.zip](https://github.com/user-attachments/files/17011519/usb-type-c-semiauto.robot_log.zip)
[usb-type-c.robot_log.zip](https://github.com/user-attachments/files/17011518/usb-type-c.robot_log.zip)

### Solutions you've tried

Rebooting & reconnecting. The v540tnd-v0.9.1-rc5-fixes branch contains a few changes to these tests which make them possible to run. The changes are not in any way related to the cause of fail afaik.

mkopec, 2024-09-17:
The logs show that the expected VIA hub device is missing, is it actually present if you run `lsusb` manually?

philipandag, 2024-09-17:
Logs after updating to current develop branch & switching to the proper docking station:
[usb-type-c.robot_log.zip](https://github.com/user-attachments/files/17029680/usb-type-c.robot_log.zip)
- I think in `UTC005.001` the usb-c dock name needs to be changed, because there is `Bus 003 Device 010: ID 2a8c:005d Sonnet Technologies, Inc. Echo 11 Thunderbolt 4 Dock`.
- I don't know about UTC005 & UTC006
- UTC013 is a PASS, the used keyboard is different
- UTC015, probably wrong device name again: Realtek Semiconductor Corp. USB Audio
- UTC018 - no SD card was inserted
- UTC019 - no SD card was inserted ???

mkopec, 2024-09-23:
Display tests fail because they look for a DP MST hub like the one in https://www.wavlink.com/en_us/product/WL-UMD05-WL-UMD05-Pro.html , but the connected dock has tunneled dual-SST displayport instead. There are separate tests for these docks now: https://github.com/Dasharo/open-source-firmware-validation/commit/2544b1c7e058768a6a16414feb1fa3e83da5a1ee

mkopec, 2024-10-03:
To be closed when @SebastianCzapla confirms the issue is gone in the next round of testing

SebastianCzapla, 2025-04-25:
More changes and fixes were made here: https://github.com/Dasharo/open-source-firmware-validation/pull/769

OSFV Dashboard results for 1.0.0-rcX releases are passing and the OSFV now distinguishes the dock types properly.

closed 2025-04-25 by SebastianCzapla