Dasharo/open-source-firmware-validation #506: V560TNE Intel ME lspci entry name is wrong

philipandag, 2024-09-17:
### Component

Dasharo firmware

### Device

NovaCustom V56 14th Gen

### Dasharo version

v0.9.1-rc4

### Dasharo Tools Suite version

--

### Test case ID

MNE002.001

### Brief summary

lspci entry with id 00:16:0 is not Intel ME

### How reproducible

100%

### How to reproduce

run `lspci | grep 00:16:0` with Intel ME enabled

### Expected behavior

should find an entry like:
`00:16:0 Intel Corporation Meteor Lake Management Engine Interface`

### Actual behavior

there is an entry:
`00:16:0 Communication controller: Intel Corporation Device 7e70 (rev20)`

### Additional context

I believe the name on the right of the device ID is determined by mapping the device ID to some table of names. This may be an error with this table in the OS, not with the ME.

### Solutions you've tried

rebooting

philipandag, 2024-09-18:
Why was it reopened if this is an issue with the OS?

mkopec, 2024-09-18:
@philipandag This is an issue that should be fixed in the tests. The tests are relying on (mutable) device names like `Management Engine Interface` that are subject to change when updating OS or when the OS does not yet know the name of the given device.

We should either have PCI VID:PID in platform configs like `8086:7e70` for the ME interface in question, or have another check that does not rely on names from the PCI ID database.

zirblazer, 2024-09-18:
> We should either have PCI VID:PID in platform configs like `8086:7e70` for the ME interface in question, or have another check that does not rely on names from the PCI ID database.

Is there any con on relying JUST on the Device ID/Vendor ID @ specific PCI Address (Not that it would change for built-in Devices, albeit I think having read on the Chipset Datasheet that some address MAY be changed) to check if Device exists? Can't think of any reason why you want to do in any other way.

philipandag, 2024-09-20:
Fixed in https://github.com/Dasharo/open-source-firmware-validation/pull/403. Works on develop branch.
[me-neuter.robot_log.zip](https://github.com/user-attachments/files/17071271/me-neuter.robot_log.zip)

closed 2024-09-20 by philipandag