Dasharo/open-source-firmware-validation #508: CBMEM003.001 wrong format of returned value results in fails

wiktormowinski, 2024-09-18:
### Device

novacustom ns70mu

### RTE version

-

### OSFV version

branch: release-v0.3.0

### Affected component(s) or functionality

`Measure Reboot Time` kwd of `lib/options/dcu.robot` resource
line 124:
```Append To List    ${durations}    ${boot_time} s```

### Brief summary

i have reasons to believe that the ,,s" in append to list kwd line is put there by mistake

### How to reproduce

try running `boot time measure` suite  

### Expected behavior

success

### Actual behavior

notice the cbmem002.001 passes while cbmem003.001 does not. In logs we can see the cbmem3 has problem with comparing values from output that is because cbmem2's list consists of float values (e. g. 2.5, 2.38, 1.3) while cbmem3 adds and extra ,,s" at the end so the returned values are no longer floats (e. g. 2.5s, 2.38s, 1.3s) As a result the test case gets confused and fails.

### Link to screenshots or logs

[logs.zip](https://github.com/user-attachments/files/17043212/logs.zip)

### Solutions you've tried

remove ,,s" from the kwd

philipandag, 2024-10-21:
Also noticed this error when testing a release. It was fixed here:
https://github.com/Dasharo/open-source-firmware-validation/pull/499/files

JakubRed, 2025-02-19:
I tested on this issue on commit [b89f7ca0472faec4e92c6e575c6cfc956c924f66](https://github.com/Dasharo/open-source-firmware-validation/commit/b89f7ca0472faec4e92c6e575c6cfc956c924f66) on main branch. There was no replication so probably the fail was caused by something else or was fixed in the meantime.

closed 2025-02-19 by macpijan