Dasharo/open-source-firmware-validation #536: MNE - Me-Neuter tests fail on MSI

JanPrusinowski, 2024-10-16:
### Device

msi-pro-z690-a-wifi-ddr4

### RTE version

1.1.4

### OSFV version

develop

### Affected component(s) or functionality

me-neuter tests

### Brief summary

Set UEFI Option is unable to set Intel ME mode

### How reproducible

100%

### How to reproduce

`robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.199 -v config:msi-pro-z690-a-wifi-ddr4 -v sonoff_ip:192.168.10.169 -v pikvm_ip:192.168.10.16 -v device_ip:192.168.10.155 -v fw_file:./msi_ms7d25_v1.1.4_ddr4.rom dasharo-security/me-neuter.robot`

### Expected behavior

Intel ME mode is set to correct value

### Actual behavior

Test fails on trying to set the option.

### Link to screenshots or logs

[image]
[image]
[log2.zip](https://github.com/user-attachments/files/17392485/log2.zip)

macpijan, 2024-10-16:
It appears that help text is being parsed, instead of the menu text.

I'd start with confirming whether it is reproducible, or if it was some one-time problem.

macpijan, 2024-10-17:
The reason of this failure is addition of these two help text lines in the selectable options area:

[image]

@BeataZdunczyk Can we try to ensure that we run some test working on certain menu when adding changes to such menus?

macpijan, 2024-10-23:
@BeataZdunczyk @JanPrusinowski So the solution would be to ignore the lines marked in yellow when parsing.

We've had something like this that checked if line was on option we could select, and dropped otherwise: https://github.com/Dasharo/open-source-firmware-validation/blob/develop/lib/bios/menus.py#L7

My suggestion would be:
- analyze the call trace - is this keyword used here?
- if no - why not, can we use it here?
- if yes - why it does not work? can we come up with better regexp?

JanPrusinowski, 2024-10-23:
The best solution would be avoiding adding any arguments to `Set UEFI Option`. Would be nice to see a solution which is automatic. In this case there is a problem because this yellow section contains the string with the option name. Maybe checking if the option exists more than once in the list would be a good approach.

macpijan, 2024-10-23:
> In this case there is a problem because this yellow section contains the string with the option name. Maybe checking if the option exists more than once in the list would be a good approach.

The linked keyword does not rely on option name to determine whether entry in menu can be selected or not.

The general rule is that if a line either:
- starts with `>`
- or has some `[]`  or `<>` later on
then this line is something we want to include in parsed menu.

miczyg1, 2024-12-16:
PR #580 merged, closing

closed 2024-12-16 by miczyg1