Dasharo/open-source-firmware-validation #537: Network Boot tests fail on MSI (duplicated output in Dasharo Network Boot Menu)

JanPrusinowski, 2024-10-16:
### Device

msi-pro-z690-a-wifi-ddr4

### RTE version

---

### OSFV version

develop

### Affected component(s) or functionality

Network Boot tests fail

### Brief summary

Duplicated output in Dasharo Network Boot Menu

### How reproducible

100%

### How to reproduce

run `robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.199 -v config:msi-pro-z690-a-wifi-ddr4 -v sonoff_ip:192.168.10.169 -v pikvm_ip:192.168.10.16 -v device_ip:192.168.10.155 -v fw_file:./msi_ms7d25_v1.1.4_ddr4_00000.rom -v capsule_fw_file:./msi_ms7d25_v1.1.4_ddr4.cap dasharo-compatibility/network-boot.robot`

login to PiKVM at: `https://192.168.10.16/kvm/`

when test will enter Dasharo Network Boot Menu - nothing will happen. Hit arrow key to aviod returning to Boot menu.

In output (in such case):
`J[0mJ[02;41HH[03;04HH[0mm[0mm[0mm      [24;04HH      [04;02HH    ------------------------------------------------  DDaasshhaarroo  NNeettwwoorrkk  BBoooott  MMeennuu  ----------------------------------------------    [78DH[05;02HH[1mm[1mm    AAuuttoobboooott  ((DDHHCCPP))                                                                                                                ((33))    [78DH[
06;02HH[0mm[0mm    DDaasshhaarroo  TToooollss  SSuuiittee                                                                                                                  [78DH[07;02HH    OOSS  iinnssttaallllaattiioonn  ((nneettbboooott..xxyyzz  ooffffiicciiaall  sseerrvveerr))                                                              [78DH[08;02HH    iiPPXXEE  SShheellll
                                                                                                             [78DH[09;02HH
 [09;02HH[10;02HH
 [10;02HH[11;02HH
 [11;02HH[12;02HH
 [12;02HH[13;02HH
 [13;02HH[14;02HH
 [14;02HH[15;02HH
    [ Message content over the limit has been removed. ]
...  iinnssttaallllaattiioonn  ((nneettbboooott..xxyyzz  ooffffiicciiaall  sse
errvveerr))                                                              [78DH[0mm[0mm    OOSS  iinnssttaallllaattiioonn  ((nneettbboooott..xxyyzz  ooffffiicciiaall  sseerrvveerr))                                                              [78DH[08;02HH[1mm[1mm    iiPPXXEE  SShheellll
                   [78DH[0mm[0mm    iiPPXXEE  SShheellll                                                                                                                                    [78DH[07;02HH[1mm[1mm    OOSS  iinnssttaallllaattiioonn  ((nneettbboooott..xxyyzz  ooffffiicciiaall  sseerrvveerr))                                                              [78DH[0mm[0mm    OOSS  iinnssttaall
llaattiioonn  ((nneettbboooott..xxyyzz  ooffffiicciiaall  sseerrvveerr))                                                              [78DH[06;02HH[1mm[1mm    DDaasshhaarroo  TToooollss  SSuuiittee                                                                                                                  [78DH[0mm[0mm    DDaasshhaarroo  TToooollss  SSuuiittee
                                                                                [78DH[05;02HH[1mm[1mm    AAuuttoobboooott  ((DDHHCCPP))                                                                                                                          [78DH
`

Each character is duplicated ... so this menu is ignored as menu option which we are looking for is altered.
However - in KVM video output everything looks correctly

### Expected behavior

Menu would be detected and most probably iPXE Shell be chosen.

### Actual behavior

Nothing happens

### Link to screenshots or logs

[image]
[log.zip](https://github.com/user-attachments/files/17394386/log.zip)

### Additional context

This looks like problems with serial duplicating output for some reason.

macpijan, 2024-10-16:
We need another RC firmware with removed iPXE serial redirection @BeataZdunczyk @SergiiDmytruk

SergiiDmytruk, 2024-10-16:
Updated https://github.com/Dasharo/coreboot/pull/509, including making CI build capsules.  As noted by @miczyg1, that's what should be used for RCs instead of `dasharo/build.sh`.  Will bump the version if it works as expected.

BeataZdunczyk, 2024-10-16:
> We need another RC firmware with removed iPXE serial redirection @BeataZdunczyk @SergiiDmytruk

We will finish testing rc1 and then gather the issues to be fixed in rc2.