Dasharo/open-source-firmware-validation #539: ESP Scanning Suite brakes Windows Boot

JanPrusinowski, 2024-10-17:
### Device

msi-pro-z690-a-wifi-ddr4

### RTE version

-

### OSFV version

develop

### Affected component(s) or functionality

ESP-Scanning Suite

### Brief summary

ESP Scanning Suite brakes Windows Boot

### How reproducible

100%

### How to reproduce

Run any test in ESP-Scanning for example:
`robot -v snipeit:no -L TRACE -v rte_ip:192.168.10.199 -v config:msi-pro-z690-a-wifi-ddr4 -v sonoff_ip:192.168.10.169 -v pikvm_ip:192.168.10.16 -v device_ip:192.168.10.155 -v fw_file:./msi_ms7d25_v1.1.4_ddr4.rom -v capsule_fw_file:./msi_ms7d25_v1.1.4_ddr4.cap -t 'ESP002*' dasharo-compatibility/esp-scanning.robot`

### Expected behavior

Suite teardown should only remove boot options added in suite prepare.

### Actual behavior

All boot options added are deleted but also files required to boot windows are removed - so boot to windows is no longer working. This breaks a lot of tests.

### Link to screenshots or logs

[log-ESP.zip](https://github.com/user-attachments/files/17421059/log-ESP.zip)

macpijan, 2024-10-23:
@JanPrusinowski Does it breaks still, even when running from this branch: https://github.com/Dasharo/open-source-firmware-validation/pull/550

JanPrusinowski, 2024-10-23:
@macpijan Did not break on provided branch:
[image]

macpijan, 2024-10-23:
So we can say it's not an issue anymore in `develop`.

closed 2024-10-23 by macpijan