Dasharo/open-source-firmware-validation #540: Support Reset To Defaults

macpijan, 2024-10-18:
**Test environment version**
* Branch:  develop
* Commit:  540916c508f3b11e3123c42c21125d4c388c1a9d

**Dasharo version (if applicable)**
 v1.1.4-rc1 for MSI zz690 DDR5

**Dasharo variant (if applicable)**

**Affected component(s) or functionality (if applicable)**
 CPU tests

**Brief summary**

CPU tests use `Reset To Defaults` kwd.

Resting firmware options causes the serial to be disabled (it is disabled by default on MSI), breaking further tests.

We have disabled CPU tests for now on these boards, but we need to see how to bring them back, if possible.

Some attempt have been made here: https://github.com/Dasharo/open-source-firmware-validation/pull/530/files#r1805154721

**Additional context**

philipandag, 2024-10-22:
If I understand correctly, the tricky part of this issue is that the `Resets To Defaults` KW can be called from any place in the setup tree, but the Serial Redirections requires us to move to a specific submenu. This could be easily solved by just rebooting the platform, but that would stop the serial redirection, so we need to check the option without reboots.

 Jan tried to solve that by creating a crawler that will try to understand where it is in the tree by scanning what is sent via serial when pressing an arrow key.  With this information it could traverse to the main menu and then update the update the SerialRedirection as usual. This approach might be difficult, have a lot of corner cases and be sensitive to changes in the setup menus.
 
 I think a much easier approach could be to reboot the platform _before_ resetting to the defaults, doing the reset in the main menu, and then, knowing we didn't leave it, check the SerialRedirection as usual.

mkopec, 2024-10-22:
UEFI has a mechanism for multiple default stores:

```
Default Stores
There are three standard default stores:
Standard Defaults — These are the defaults used to prepare the system/device for normal operation.
Manufacturing Defaults — These are the defaults used to prepare the system/device for manufacturing.
Safe Defaults — These are the defaults used to boot the system in a “safe” or low-risk mode.
```

Maybe add an option to reset to manufacturing defaults and ensure that the manuf default for serial redirection is `TRUE`?

philipandag, 2024-10-22:
That would require actually configuring these stores, right? If we are not doing it already in the release binaries then that would be a pretty invasive change. If we do then there's still the issue how to access them. I haven't seen anything like that in the UEFI Setup menu. The DCU does not expose variables related to this either. Wouldn't it require rebuilding the FW? Maybe you meant to add an option to reset do manuf defaults to edk2?

That might be the best, but much more complex option. This would require updating all the devices which support the CPU tests or determining if the manuf settings are available during the test case...

philipandag, 2024-10-22:
The argument against always performing a reboot before restoring to the defaults is that the  reset to defaults changes the options in setup menu immediately. To commit them a reboot is needed but the defaults can be read by traversing the setup menu. 

Some tests rely on this by entering a submenu, changing an option, performing a restore to defaults and checking if the option has changed immediately, knowing that the option tested is still visible. Performing a reboot before this would might break these tests.

philipandag, 2024-10-24:
I wonder if just pressing "ESC" repeatedly, like 10 times, won't be enough to return to the main Setup menu. One place where it would surely fail is when there is an error in applying a form and the user is asked to, if I remember correctly, either press "d" to revert the changes or something else to return to the form. Although in this case any automated test in Setup menu would fail as it is an unexpected situation.

krystian-hebel, 2024-11-12:
> I wonder if just pressing "ESC" repeatedly, like 10 times, won't be enough to return to the main Setup menu.

It would get to top level menu, but it wouldn't move the cursor. This problem can be handled by pressing Page Up, it always moves to first entry because there is no scrolling on the front page (at least currently).

It may be possible to add a new hotkey to return to top level immediately in edk2. Alternatively we may modify F9 to return to top as well. Both options would require changes to edk2 code and this may be an overkill.

> One place where it would surely fail is when there is an error in applying a form and the user is asked to, if I remember correctly, either press "d" to revert the changes or something else to [...]

Agree, tests should be written to catch errors, not to work around them.

philipandag, 2024-11-19:
The `Reset To Defaults Tianocore` keyword is always used before `Save Changes And Reset`.
There was an argument against performing a reboot to get back to the main setup screen
> Some tests rely on this by entering a submenu, changing an option, performing a restore to defaults and checking if the option has changed immediately, knowing that the option tested is still visible. Performing a reboot before this would might break these tests.

But it's not true. The `Reset To Defaults Tianocore` keyword could easily be changed so that it reboots, performs the edk2 reset to defaults and restores the serial redirection from there without caring about traversing the setup menu.

What do you think @miczyg1

philipandag, 2024-11-19:
https://github.com/Dasharo/open-source-firmware-validation/pull/601

I have modified the Reset To Defaults Tianocore keyword to always set the serial redirection to Enabled after resetting as suggested in https://github.com/Dasharo/open-source-firmware-validation/pull/530/files#r1805158430

Running the CCC test on MSI Z690 DDR5 with i5 13600k shows that  the platform still has the serial redirection enabled after restoring to deefaults. After CCC007 the platform became unresponsive though and required a hard reset which I don't think could be caused by the change. It could use a rerun but I don't know if that's in the scope of this issue.

[cpu-cores-count.robot_log.zip](https://github.com/user-attachments/files/17815673/cpu-cores-count.robot_log.zip)

miczyg1, 2024-11-21:
> The `Reset To Defaults Tianocore` keyword is always used before `Save Changes And Reset`. There was an argument against performing a reboot to get back to the main setup screen
>
>
> But it's not true. [...]

Okay, I might have remembered some old test version. However, I'm no convinced it is that easy to change. All tests perform reset to defaults then reset a platform. So before platform reset happens, you would have to navigate to Dasharo System Features -> Serial Console Redirection submenu. However, you don't know which submenu you are currently in. You would have to traverse the menu back to main page and then to  Dasharo System Features -> Serial Console Redirection. I heard many times that going forward in menus is not a problem, but going backwards is not trivial. As long as you can implement it reliably, I'm fine with that.

philipandag, 2024-11-21:
>  All tests perform reset to defaults then reset a platform. So before platform reset happens, you would have to navigate to Dasharo System Features -> Serial Console Redirection submenu. 

I think that would only be the case if resetting the serial redirection option was tested, but it is not. Are there any test cases that depend on the Reset To Defaults Tianocore to return from an invalid configuration, booting which should be avoided?

philipandag, 2024-11-21:
The reason for the failures in the CCC test suite was indeed the `Power On` keyword. After increasing the sleep durations in it the CCC test suite passes perfectly
[cpu-cores-count.robot_log.zip](https://github.com/user-attachments/files/17846203/cpu-cores-count.robot_log.zip)

miczyg1, 2024-11-22:
> I think that would only be the case if resetting the serial redirection option was tested, but it is not. Are there any test cases that depend on the Reset To Defaults Tianocore to return from an invalid configuration, booting which should be avoided?

Resetting to default causes all settings to be reset. On the example of MSI, resetting to defautls will disabel serial consoel access, because the default is disabled.  So `Reset To Defaults Tianocore` does the job right, btu the test are unable to handle the loss of serial output. Is that more clear now?