Dasharo/open-source-firmware-validation #564: CBO Custom Boot Order test should be implemented

philipandag, 2024-11-05:
### The problem you're addressing (if any)

The CBO test suite is not implemented in osfv despite being shown as automated on the test matrix. I can't see why it couldn't be implemented, although it might be a bit complex. Custom boot order can be modified and checked in the setup menu via serial or in the OS using efibootmgr. 

### Describe the solution you'd like

The test case should be implemented in osfv

### Where is the value to a user, and who might that user be?

Testers not having to do it manually

### Additional context

Doing the test manually doesn't take that much time so it might not be worth it.

macpijan, 2025-01-10:
`CBO001.002 Custom boot order (edk2)` was identified as missing in automation during odroid validation.

We could add it to backlog if we get a chance to implement some new tests @BeataZdunczyk

SebastianCzapla, 2025-04-28:
Fixed in https://github.com/Dasharo/open-source-firmware-validation/pull/812

closed 2025-04-29 by filipleple