Dasharo/open-source-firmware-validation #565: TPM Detect tests don't support TPM 1.2

philipandag, 2024-11-05:
### The problem you're addressing (if any)

TPD tests use tpm2-tools which obviously only work with TPM 2.0. Performing the tests on Optiplex 7010 fails because it has TPM 1.2.

### Describe the solution you'd like

Provide an alternative way of detecting TPM if using tpm2-tools won't work.

### Where is the value to a user, and who might that user be?

Otherwise the tests can't be performed on devices with older TPMs 

### Describe alternatives you've considered

instead of tpm2-tools maybe one of the following could be used?

dmesg | grep tpm
journalctl -g tpm
ls /sys/class/tpm
cat /sys/class/tpm/tpm*/tpm_version_major

mkopec, 2024-11-05:
> instead of tpm2-tools maybe one of the following could be used?

yes that would work if it's sufficient for tests

there's also tpm-tools from the TrouserS project

philipandag, 2024-11-19:
I think this PR https://github.com/Dasharo/open-source-firmware-validation/pull/487/files included a keyword that could be used in those tests.

SebastianCzapla, 2025-02-04:
https://github.com/Dasharo/open-source-firmware-validation/pull/507 was merged

closed 2025-02-04 by SebastianCzapla