Dasharo/open-source-firmware-validation #567: Errors in ESP scan tests

miczyg1, 2024-11-07:
### Device

Any

### RTE version

Any

### OSFV version

Good question

### Affected component(s) or functionality

ESP scanning tests

### Brief summary

The keywords in teardown and ESP005.001 have bugs

### How to reproduce

Run the test

### Expected behavior

No typos in commands

### Actual behavior

Potential test fails and wiped Windows Boot Manager from ESP

### Link to screenshots or logs

https://github.com/Dasharo/dasharo-issues/issues/1128#issuecomment-2462011928
https://github.com/Dasharo/dasharo-issues/issues/1128#issuecomment-2462023366

JanPrusinowski, 2024-11-07:
This is related to: https://github.com/Dasharo/open-source-firmware-validation/issues/539

SebastianCzapla, 2025-04-17:
Issue no longer relevant, the problems mentioned in comments have been changed during other efforts, for example, switching from cloud DL to osfv-test-data repo.

closed 2025-04-17 by SebastianCzapla