Dasharo/open-source-firmware-validation #583: Add more details to robot docs

EduKav1813, 2024-11-14:
### The problem you're addressing (if any)

The current version of the robot documentation does not provide all necessary information that may help to develop or debug tests.

### Describe the solution you'd like

Add the following information to the robot documentation:

1. Type hints for the arguments. If I see something like: Arguments: item it would help to know what type it is.
2. The type hints for the return values. The same as in the previous one.
3. Not all keywords have Documentation. Even if their name may be self-descriptive, it could confuse new users.

### Where is the value to a user, and who might that user be?

Provide more relevant context for test developers.

philipandag, 2024-12-17:
>  some side effects of using keyword e.g. whether keyword needs to be used inside Linux shell or during booting

I totally agree. At this moment using the keywords pretty much requires first looking up how they work inside. As for this, I believe the only good fix would be to create a guideline somewhere in the OSFV repo, tell everyone to use it for new keywords and give someone the task of fixing the documentation for every existing keyword, test case and test suite according to the guidelines.

> Not all keywords have Documentation.

There is a robocop check for that, just that it is being skipped for now because... not all keywords have documentation
```
    # TODO: add missing documentation for keywords and re-enable
    "0201:enabled:False",
```` 
There are many more checks skipped there, most marked with something like `#TODO enable after fix`.

philipandag, 2024-12-17:
> Type hints for the arguments. If I see something like: Arguments: item it would help to know what type it is.

Adding type annotations to keywords defined using RF is not supported right now with RF 7.0.1. There is an issue in their repository for this feature: https://github.com/robotframework/robotframework/issues/3278.

As for now we could just require to manually say in the documentation what types are considered valid for a given argument.

philipandag, 2024-12-17:
I'm working on creating the documentation guidelines here:
https://github.com/Dasharo/open-source-firmware-validation/pull/new/rf-documentation-guidelines

I have proposed a template for KW documentation. It's hard to not notice that it is a bit long compared to existing documentations. Writing complex in-code documentation could negatively affect the process of creating and maintaining the keywords.

The documentation of test cases and test suites could probably also use some standarization and/or extension. I am not sure what we would wan't to include in them though. @mkopec Would you mind to share what you think?

m-iwanicki, 2024-12-17:
@philipandag There is keyword length limit of maybe 50 lines in our pre-commit? I'm not sure if it's https://github.com/Dasharo/open-source-firmware-validation/blob/b89f7ca0472faec4e92c6e575c6cfc956c924f66/robocop.toml#L26 but from what I remember comments counted for this limit. 
If it still does then we need to at least increase this limit or maybe add such a feature to `robocop`.

philipandag, 2024-12-17:
> @philipandag There is keyword length limit of maybe 50 lines in our pre-commit? I'm not sure if it's

I am aware of that and also thought that it might pose a problem. Defining keywords and their documentation in Python would certainly not trigger this robocop check, what's more, defining type hints for arguments and return values is possible if the keyword is implemented in Python. Rewriting most keywords in Python wouldn't be a good idea though, so I guess we need to document the KWds in RF.

Do you think such verbose documentation as in https://github.com/Dasharo/open-source-firmware-validation/pull/637 i a good idea?

philipandag, 2025-01-09:
Robot Framework suggests not creating documentation, or making it really short if absolutely needed.

> If the logic of the test case needs documenting, it is often a sign that keywords in the test case need better names and they are to be enhanced, instead of adding extra documentation.

~ RobotFrameworkUserGuide

That would be possible in an ideal scenario. I doubt our keywords could be exhaustively described without precise documentation...

m-iwanicki, 2025-01-09:
> If the logic of the test case needs documenting

I think it's about test cases not keywords.

> Can you say more specifically what is the problem with the indentation

I think multiline bulletpoints look very bad, I don't think it treats it as bullet point continuation but just new text in new line. Should probably follow markdown indentation:

```markdown
**Side effects**
* Creates `${out_file}` file with the cowsay message. Overwrites
the file if already exists.
```

**Side effects**
* Creates `${out_file}` file with the cowsay message. Overwrites
the file if already exists.

philipandag, 2025-01-09:
I see. The interline spacing is off. Changing the indentation does not affect it though, and the documentation strings do not use markdown syntax. The lists are rendered properly only if the whole text is in one line.

Does this look good to you?
[image]

m-iwanicki, 2025-01-09:
Based on documentation: http://robotframework.org/robotframework/latest/RobotFrameworkUserGuide.html#lists
it should work but:
https://github.com/robotframework/robotframework/issues/3230

>  if documentation needs to be updated, we can open a separate issue
5 years and still the same, maybe it was fixed in newer robot framework versions but currently you need to add `\ ` e.g.:

```
...    - Creates `${out_file}` file with the cowsay message. Overwrites
   ...    \ the file if already exists.
```

philipandag, 2025-01-14:
The documentation is not wrong in the end. It's just that we are using a RF version from nearly three years ago. The documentation mentions in one place that spaces need to be espaced in version pre 6.0 IIRC, but does not repeat this again.

m-iwanicki, 2025-01-14:
@philipandag same thing with RF 7.1.1. It's probably related more to using spaces as argument separator?

philipandag, 2025-01-14:
@m-iwanicki After upgrading RF and libdoc to 7.1.1 it worked for me just like in the documentation. Are you sure the version of the used executables is 7.1.1?

m-iwanicki, 2025-01-14:
```
(venv) miwanicki@3M02 ➜  ~/projects/open-source-firmware-validation/scripts git:(run-dts-tests-faster) ✗ cat tmp.robot
*** Keywords ***
Cowsay Keyword
    [Documentation]
    ...    === Return Value ===
    ...    - ``string`` - The generated cowsay message
    ...    \ correct indentation
    ...
    ...    === Side Effects ===
    ...    - Creates ``${out_file}`` file with the cowsay message. Overwrites
    ...    the file if already exists.
    [Arguments]    ${input_text}    ${out_file}="cowsay.txt"
    Execute Command In Terminal    cowsay ${input_text} > ${out_file}
```

```
$ libdoc tmp.robot tmp.html
~/projects/open-source-firmware-validation/scripts/tmp.html
$ which libdoc
~/projects/open-source-firmware-validation/scripts/venv/bin/libdoc
$  robot --version
Robot Framework 7.1.1 (Python 3.13.0 on linux)
```

m-iwanicki, 2025-01-14:
Deleted previous comment, forgot to add additional indentation space

mkopec, 2025-01-14:
@philipandag left you a comment under both PRs. Otherwise the guidelines look good. I'm not sure about the scope though, going through every single kwd may be too much. Let's start with the most used keywords

philipandag, 2025-01-14:
Keyword hermetization is being implemented in RF, although it won't appear in RF7. It might be very useful in osfv. The syntax for marking keywords as private is pretty much decided upon, so I suggest we mark local keywords as private.

All that does now is produce a warning. The keyword can still be used and the robot LSP from robocorp still suggests them in autocompletions. That might change in the future.

philipandag, 2025-01-14:
I think the first batch of keywords to document should include:
- Power On - it behaves differently depending on device, it should be documented why. I am not sure how to document it as it is redefined multiple times in the platform configs. Maybe we could make a dummy keyword which contains the documentation and is overridden by the implementations from platform configs?
- whole lib/bios/menus
- whole lib/options - the same issue as with `Power On`, duplicating the documentation on every definition is not ideal
- lib/terminal
- Power Cycle On - why it's different from Power On
- multiple reboot kwds

EduKav1813, 2025-01-15:
@philipandag cc

m-iwanicki, 2025-01-15:
Ad. 4: I think might need changes in libdoc code, at least I couldn't find any way to add argument description to automatically generated argument section
Ad. 5: I don't agree, we return values not variables.

philipandag, 2025-01-16:
> Documentation and Description are the same in this case. I would try to make keyword documentation more compact by removing unnecessary things.

Don't you think a one-sentence description of what the keyword does is good to have? It is not absolutely necessary, albeit I don't agree it's useless.

Or maybe you meant to remove the section title? That would make more sense and with that I agree.

> As @m-iwanicki said, indentation. It should be easy to read, so we should either replace bulletpoints, or (if we have to) manually indent it in description.

What would you prefer over bullet points? I think Michał only had an issue with them because the formatting of multiline bullet-points was broken. Do you think writing full sentences would be better?

> Arguments is duplicated, I would keep the later one.

As Michał said, the first one is automatically generated from the keyword definition. I'll check if there is a way to remove it or include type hints / description in the automatically generated section.

> the person is experienced and then it doesn't matter, or inexperienced, and then this doesn't really provide much new information. If anyone has any hints on this, I would be happy to hear them.

Isn't the whole point of documentation to allow inexperienced people to use the code?

For example let's take a look at the keyword "Enter Boot Menu Tianocore And Return Construction".
What is a "Construction"? I think it would be beneficial to say that it's just a simple string containing the lines of the boot menu, seperated with newlines, with most "decorative" lines removed.

philipandag, 2025-01-16:
I don't think there is a proper way to add more details or remove the automatically generated Arguments section. It could be removed in post process if it looks bad...

philipandag, 2025-01-16:
> I think the first batch of keywords to document should include:
> 
>     * Power On - it behaves differently depending on device, it should be documented why. I am not sure how to document it as it is redefined multiple times in the platform configs. Maybe we could make a dummy keyword which contains the documentation and is overridden by the implementations from platform configs?
> 
>     * whole lib/bios/menus
> 
>     * whole lib/options - the same issue as with `Power On`, duplicating the documentation on every definition is not ideal
> 
>     * lib/terminal
> 
>     * Power Cycle On - why it's different from Power On
> 
>     * multiple reboot related kwds in keywords.robot

The behavior for the `Power On` keyword can be achieved using the `Set Library Search Order` built-in keyword. By default all resource file keywords are treated the same, but we can use `Set Library Search Order    ${CONFIG}` in the `Prepare Test Suite` to allow overwriting any keyword in the platform configs.

This way the `Power On` keyword will be known by code completion tools and will be included in the documentation. We can define it in `keywords.robot` such that it causes a `Fail` if used without a proper implementation in the config.

Being able to override keywords in platform configs has many pros, but it could lead to a mess if overused.