Dasharo/open-source-firmware-validation #606: Upgrade to RF7

macpijan, 2024-11-21:
- Possibly resolving limitation mentioned in: https://github.com/Dasharo/open-source-firmware-validation/pull/501#discussion_r1852073048

pekkaklarck, 2024-11-22:
Upgrading at least to RF 6.1 gives you Python 3.12 compatibility. Without it the `$var` syntax doesn't work:
https://github.com/robotframework/robotframework/issues/4771

Upgrading to RF 7.1 gives official Python 3.13 support, but we didn't actually need to make any code changes to support it so also earlier versions are compatible:
https://github.com/robotframework/robotframework/issues/5091

macpijan, 2024-11-27:
There is a new thread regarding switch to this lib, that it must be updated to telnetlib3, as telnetlib got removed with Python 3.13: https://docs.python.org/3/library/telnetlib.html

SergiiDmytruk, 2024-11-28:
Not exactly new, it was mentioned in some relevant issues.  `exscript` module is another option and looks more like `telnetlib`, because `telnetlib3` seems to be asynchronous which wouldn't be used in RF anyway (I assume there is a way to force async functions to be synchronous if needed).

macpijan, 2024-11-28:
Thanks for this useful background

macpijan, 2025-07-09:
No activity here means no status update.

There was no activity around this as this was not strictly necessary so far.

DaniilKl, 2025-07-09:
Please, consider scheduling the update.

closed 2025-07-29 by macpijan