Dasharo/open-source-firmware-validation #608: Login To Linux keyword sometimes fails over serial console

SebastianCzapla, 2024-11-22:
### Device

-

### RTE version

-

### OSFV version

Develop branch, latest commit 2923c8f7da762b26c52e9853b2eca8751dc98de8

### Brief summary

Keword times-out

### How reproducible

100%

### How to reproduce

Run dasharo-compatibility/efibootmgr.robot

### Expected behavior

Tests run or skip properly depending on the DUT

### Actual behavior

Suite hangs on setup/teardown

### Link to screenshots or logs

[image]

### Additional context

Login To Linux keyword waits for "login:" string. This has failed due to unclear state of the DUT after other suites.
For example, when run with other suites, "login:" could not be found, however manual login to machine shows that it is already logged in with ubuntu user.

Even if user is not logged, connecting via telnet shows no prompt, unless machine is actively booting and we connect just in time to catch "login:" prompt.

macpijan, 2024-11-22:
@SebastianCzapla If you analyze what this keyword does, you will notice:
1. It enables power supply via sonoff (at this point Protectli already starts, as it boots on power applied)
2. It pressess the power buttin - powering down the unit.

The following diff lets this first test to be executed:

```
diff --git a/lib/options/uefi-setup-menu.robot b/lib/options/uefi-setup-menu.robot
index 0a5d056ee00e..ebd6642227e4 100644
--- a/lib/options/uefi-setup-menu.robot
+++ b/lib/options/uefi-setup-menu.robot
@@ -86,7 +86,7 @@ Measure Coldboot Time
     Log To Console    \n
     # Do one more iteration than requested, as we may hit first boot which is always longer.
     FOR    ${index}    IN RANGE    0    ${iterations}+1
-        Power Cycle On    power_button=${TRUE}
+        Power Cycle On
         Boot System Or From Connected Disk    ubuntu
         Login To Linux
         Switch To Root User
```

I think to properly resolve it, we need to implement the support of achieving desired power state in the osfv_cli: https://github.com/Dasharo/osfv-scripts/issues/69

macpijan, 2024-11-22:
```
erformance/boot-time-measure.robot
==============================================================================
Boot-Time-Measure                                                             
==============================================================================
qCBMEM001.001 Serial boot time measure: coreboot booting time after... ..

q(0) Boot time: 1.453645 s
(1) Boot time: 1.448787 s
(2) Boot time: 1.453494 s
(3) Boot time: 1.452893 s
(4) Boot time: 1.450988 s
(5) Boot time: 1.448356 s
.
Coreboot average booting time: 1.4509035999999997 s

.
Coreboot shortest booting time: 1.448356 s

.
Coreboot longest booting time: 1.453494 s

.
Coreboot booting time std dev: 0.002080637844508275 s

CBMEM001.001 Serial boot time measure: coreboot booting time after... | PASS |
```

SebastianCzapla, 2025-02-20:
Fixed in https://github.com/Dasharo/open-source-firmware-validation/pull/698

SebastianCzapla, 2025-04-08:
Closing, as the fix was merged and issue no longer occurs.

closed 2025-04-08 by SebastianCzapla